Edzelf/ESP32-Radio #275: Adafruit_7735 bluetft.h help

mayra6385, 2019-09-29:
Can anyone, please, help me with this isuue?

Arduino: 1.8.10 (Windows 7), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

Multiple libraries were found for "PubSubClient.h"
In file included from D:\MAYRA FILES\My Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library/Adafruit_ST7735.h:4:0,

 Used: D:\MAYRA
                 from sketch\bluetft.h:9,

Multiple libraries were found for "SD.h"
 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SD
 Not used: C:\Program
                 from C:\Users\M-Komp\Desktop\ESP32-Radio-master\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino:1072:

Multiple libraries were found for "ArduinoOTA.h"
 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\ArduinoOTA
Multiple libraries were found for "Update.h"
 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\Update
Multiple libraries were found for "WiFiMulti.h"
 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi
Multiple libraries were found for "ESPmDNS.h"
D:\MAYRA FILES\My Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library/Adafruit_ST77xx.h:30:26: fatal error: Adafruit_GFX.h: No such file or directory

 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\ESPmDNS
Multiple libraries were found for "FS.h"
 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\FS
Multiple libraries were found for "SPI.h"
compilation terminated.

 Used: C:\Users\M-Komp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPI
Multiple libraries were found for "Adafruit_ST7735.h"
 Used: D:\MAYRA
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

moresavanyu, 2019-10-06:
I have the same problem , at least my log says it is something with adafruit library. i would appreciate some help.
C:\Users\moresavanyu\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library/Adafruit_ST77xx.h:30:26: fatal error: Adafruit_GFX.h: No such file or directory

mayra6385, 2019-10-06:
Found solution! Include this library https://github.com/adafruit/Adafruit-GFX-Library