Edzelf/ESP32-Radio #277: Multiple libraries when using Arduino IDE

hoahoctro, 2019-10-27:
Arduino IDE shows this list of errors when compiling, there're too many multiple libs so that i dont know which should be deleted or kept
I searched for the lib conflict problems but cant find a good solution, please help

> Arduino: 1.8.10 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 512000, None"

Multiple libraries were found for "PubSubClient.h"
In file included from C:\Users\DELL\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library/Adafruit_ST7735.h:4:0,

 Used: C:\Users\DELL\Documents\Arduino\libraries\PubSubClient
                 from sketch\bluetft.h:9,

Multiple libraries were found for "WiFiMulti.h"
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
Multiple libraries were found for "ESPmDNS.h"
                 from C:\Users\DELL\Documents\Arduino\libraries\ESP32-Radio\Esp32_radio\Esp32_radio.ino:1072:

 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS
C:\Users\DELL\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library/Adafruit_ST77xx.h:30:26: fatal error: Adafruit_GFX.h: No such file or directory

Multiple libraries were found for "SD.h"
compilation terminated.

 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "Update.h"
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update
Multiple libraries were found for "FS.h"
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
Multiple libraries were found for "SPI.h"
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Multiple libraries were found for "ArduinoOTA.h"
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA
Multiple libraries were found for "Adafruit_ST7735.h"
 Used: C:\Users\DELL\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Edzelf, 2019-10-28:
Most of the libraries are part of the ESP32 toolkit.  Here is the complete list:
```
PubSubClient at version 2.7 in folder: C:\Users\Ed\Documents\Arduino\libraries\PubSubClient 
WiFi at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
ESPmDNS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
FS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
SD at version 1.0.5 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD 
SPI at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
ArduinoOTA at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA 
Update at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update 
Adafruit_ST7735_Library at version 1.0.8 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_ST7735_Library 
Adafruit_GFX_Library at version 1.5.7 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_GFX_Library 
```

hoahoctro, 2019-10-28:
So i must delete these libs ?. But if, then i cant use them for Arduino boards ( or must reinstall)

Edzelf, 2019-10-28:
First try to add gfx library.  That's what the error message says.

hoahoctro, 2019-10-28:
Thanks. But hows about nvs.h not found ? . Should i use ArduinoNVS ?