Edzelf/ESP32-Radio #279: ESP32 + ILI9341 Wiring Diagram

AlexVichter, 2019-11-09:
Dear Ed,
your Web Radio ESP32 project supports the ILI9341 display. I can find in your description only the wiring diagram for the Display with the following pins:
D/C (A0),
CLK/SCK,
DIN/SDA
CS
GND
BL
VCC
RST pins

But my display 3.2 TFT SPI 240x320 ILI9341 has another pins:
SDO(MISO)
LED
SCK
SDI(MOSI)
DC
RESET
CS
GND
VCC
http://www.lcdwiki.com/images/thumb/c/c4/MSP3218-021.jpg/500px-MSP3218-021.jpg

Can you please advise the wiring diagram for this variant to ESP32 device according to the actual .ino sketch?

Should I change any variables in your .ino sketch to adopt my ILI9341? If yes, which file and which lines in it?

Perhaps my questions are stupid, but I cannot find this information in project files.

Thanks a lot for your help and support!

Edzelf, 2019-11-10:
No need to change the software.  It's just a matter of different names.. Use this translation:
```
SDO(MISO)	<not used>
LED		BL
SCK		CLK/SCK
SDI(MOSI)	DIN/SDA
DC		D/C (A0)
RESET		RST
CS		CS
GND		GND
VCC		VCC
```

AlexVichter, 2019-11-10:
Thank you Ed!