Edzelf/ESP32-Radio #281: volume goes down to zero automatically

TCakshay, 2019-11-26:
i have tried the setup without connecting a display .it works fine ,but volume goes down to zero with no reason

Edzelf, 2019-11-26:
If the program turns the volume down, you should see the reason in the logging on the serial port.

TCakshay, 2019-11-26:
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Publish to topic PF9BF713C/volume : 51
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Publish to topic PF9BF713C/volume : 34
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Publish to topic PF9BF713C/volume : 13
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1

TCakshay, 2019-11-26:
this is the serial output that i get while its streaming

Edzelf, 2019-11-26:
Notice that that the program sees left rotation of the rotary encoder, which is interpreted as "lower the volume".  Find out what pins are used for the encoder.  It is probably not configurated....

TCakshay, 2019-11-26:
yeah, i have got it. thank you

closed 2019-11-26 by TCakshay