Edzelf/ESP32-Radio #283: SPI semaphore VS1053 no sound

minzukas, 2019-12-16:
One more to no output sound on the VS1053.
All connected as per instructions. WEB interface seems to work but no sound on headphones.
Tries to read the reported issues however that did not solve my problem. DummyTFT in use as i only have hx1230 display at the moment.

Log:

02:43:49.272 -> ⸮r⸮l⸮|⸮$⸮⸮⸮⸮⸮⸮$⸮⸮,:l⸮;⸮"#⸮$⸮⸮⸮⸮$d$:⸮⸮⸮$8⸮⸮''⸮⸮⸮#$⸮l⸮⸮⸮⸮⸮⸮l⸮⸮$ ?⸮$⸮$;$⸮'&⸮⸮;'#b⸮⸮l⸮⸮⸮⸮⸮`⸮nr⸮⸮b⸮""⸮⸮⸮b⸮bb⸮⸮|⸮bbp⸮⸮⸮<⸮b#'⸮⸮<⸮""⸮⸮⸮<⸮#$ ;⸮'⸮⸮⸮⸮⸮>⸮#⸮⸮'&pn⸮⸮bll`p⸮n⸮br⸮⸮⸮⸮$"'⸮<"⸮$0⸮'⸮";⸮⸮⸮⸮$⸮⸮#'⸮<#$⸮⸮$8⸮'⸮b⸮⸮bn⸮|bl⸮⸮lp⸮n⸮b⸮⸮#'⸮|bl⸮l⸮⸮l⸮n⸮<⸮⸮⸮⸮⸮⸮`
02:43:49.508 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 289148
02:43:49.508 -> D: Display type is DUMMYTFT
02:43:49.508 -> D: Partition nvs found, 20480 bytes
02:43:49.508 -> D: Read 44 keys from NVS
02:43:49.508 -> D: pin_ir set to 35
02:43:49.508 -> D: pin_enc_clk set to 25
02:43:49.508 -> D: pin_enc_dt set to 26
02:43:49.534 -> D: pin_enc_sw set to 27
02:43:49.534 -> D: pin_tft_cs set to 15
02:43:49.534 -> D: pin_tft_dc set to 2
02:43:49.534 -> D: pin_tft_scl set to -1
02:43:49.534 -> D: pin_tft_sda set to -1
02:43:49.534 -> D: pin_tft_bl set to -1
02:43:49.534 -> D: pin_tft_blx set to -1
02:43:49.534 -> D: pin_sd_cs set to 21
02:43:49.534 -> D: pin_vs_cs set to 5
02:43:49.534 -> D: pin_vs_dcs set to 16
02:43:49.567 -> D: pin_vs_dreq set to 4
02:43:49.567 -> D: pin_shutdown set to -1
02:43:49.567 -> D: pin_shutdownx set to -1
02:43:49.567 -> D: pin_spi_sck set to 18
02:43:49.567 -> D: pin_spi_miso set to 19
02:43:49.567 -> D: pin_spi_mosi set to 23
02:43:49.567 -> D: GPIO0 is HIGH
02:43:49.567 -> D: GPIO2 is LOW, probably no PULL-UP
02:43:49.567 -> D: GPIO4 is HIGH
02:43:49.567 -> D: GPIO5 is HIGH
02:43:49.827 -> D: GPIO12 is HIGH
02:43:49.827 -> D: GPIO13 is HIGH
02:43:49.827 -> D: GPIO14 is HIGH
02:43:49.827 -> D: GPIO15 is HIGH
02:43:49.827 -> D: GPIO16 is HIGH
02:43:49.827 -> D: GPIO17 is HIGH
02:43:49.827 -> D: GPIO18 is HIGH
02:43:49.827 -> D: GPIO19 is HIGH
02:43:49.827 -> D: GPIO21 is HIGH
02:43:49.827 -> D: GPIO22 is HIGH
02:43:49.827 -> D: GPIO23 is HIGH
02:43:49.827 -> D: GPIO25 is HIGH
02:43:49.827 -> D: GPIO26 is HIGH
02:43:49.827 -> D: GPIO27 is HIGH
02:43:49.827 -> D: GPIO32 is HIGH
02:43:49.827 -> D: GPIO33 is HIGH
02:43:49.827 -> D: GPIO34 is LOW, probably no PULL-UP
02:43:49.827 -> D: GPIO35 is LOW, probably no PULL-UP
02:43:49.827 -> D: GPIO39 is LOW, probably no PULL-UP
02:43:49.827 -> D: gpio_00 will execute uppreset = 1
02:43:49.827 -> D: gpio_12 will execute upvolume = 2
02:43:49.827 -> D: gpio_13 will execute downvolume = 2
02:43:49.827 -> D: Enable pin 35 for IR
02:43:49.827 -> D: Start display
02:43:50.197 -> D: SD Card Mount Failed!
02:43:50.197 -> D: Create list with acceptable WiFi networks
02:43:50.197 -> D: Added XXXXX to list of networks
02:43:50.197 -> D: Added SSID2 to list of networks
02:43:50.197 -> D: End adding networks
02:43:50.362 -> D: Scan Networks
02:43:52.685 -> D: Scan completed
02:43:52.685 -> D: Number of available networks: 8
02:43:52.685 -> D:  1 - Balticum12-2              Signal: -70 dBm, Encryption WPA_WPA2_PSK, 
02:43:52.685 -> D:  2 - Rubuilijus                Signal: -81 dBm, Encryption WPA_WPA2_PSK, 
02:43:52.685 -> D:  3 - NETGEAR35                 Signal: -81 dBm, Encryption WPA2_PSK, Acceptable
02:43:52.718 -> D:  4 - PS4-B6645C2DBBC4          Signal: -83 dBm, Encryption WPA2_PSK, 
[... 273 lines cut ...]
02:44:03.562 -> D:   1 -    33
02:44:03.562 -> D: SPI semaphore not taken within 1930 ticks by CPU 0, id hspec
02:44:03.562 -> D:   2 -     0
02:44:03.562 -> D:   3 -  6000
02:44:03.595 -> D: SPI semaphore not taken within 1940 ticks by CPU 0, id hspec
02:44:03.595 -> D:   4 -     0
02:44:03.595 -> D:   5 -  AC45
02:44:03.595 -> D: SPI semaphore not taken within 1950 ticks by CPU 0, id hspec
02:44:03.595 -> D:   6 -  1800
02:44:03.595 -> D:   7 -  1E06
02:44:03.628 -> D: SPI semaphore not taken within 1960 ticks by CPU 0, id hspec
02:44:03.628 -> D:   8 -     0
02:44:03.628 -> D:   9 -     0
02:44:03.628 -> D: SPI semaphore not taken within 1970 ticks by CPU 0, id hspec
02:44:03.628 -> D:   A -     0
02:44:03.628 -> D:   B -  F8F8
02:44:03.628 -> D: SPI semaphore not taken within 1980 ticks by CPU 0, id hspec
02:44:03.628 -> D:   C -     0
02:44:03.628 -> D:   D -     0
02:44:03.628 -> D: SPI semaphore not taken within 1990 ticks by CPU 0, id hspec
02:44:03.628 -> D:   E -     0
02:44:03.628 -> D:   F -     0
02:44:04.159 -> D: Metadata block 64 bytes
02:44:04.159 -> D: Streamtitle found, 53 bytes
02:44:04.159 -> D: StreamTitle='NPO Radio 1 - Zwarte Prietpraat - POWN';

Edzelf, 2019-12-16:
I tried to reproduce this error, but without success.  I have changed the claimSPI() function to allow some more debug logging.  Could you try this new version?

minzukas, 2019-12-16:
Thanks, will try it after the work

minzukas, 2019-12-16:
here it is with the new code:

19:29:07.368 -> ⸮ ⸮ ⸮r⸮l⸮|⸮l⸮⸮⸮⸮⸮⸮l⸮⸮lrl⸮r⸮bb⸮l⸮⸮⸮⸮lllr⸮⸮⸮lp⸮⸮n'⸮⸮⸮#$⸮$⸮⸮⸮⸮⸮⸮$⸮⸮$ ?⸮$⸮$;$⸮''⸮⸮;'#b⸮⸮l⸮⸮⸮⸮⸮`⸮nr⸮⸮b⸮bb⸮⸮⸮b⸮bb⸮⸮|⸮b#8⸮⸮⸮<⸮bbn⸮⸮<⸮##⸮⸮⸮<⸮#$ ;⸮'⸮⸮⸮⸮⸮~⸮b⸮onpn⸮⸮⸮"$$ 8⸮'⸮":⸮⸮⸮⸮$#'⸮<#⸮$8⸮'⸮#;⸮⸮⸮⸮l⸮bn⸮|bl⸮⸮$8⸮'⸮"⸮⸮⸮#n⸮|bl⸮⸮lp⸮n⸮#⸮⸮bn⸮|bl⸮l⸮⸮l⸮n⸮|⸮⸮⸮⸮⸮⸮`
19:29:07.550 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 Dec 2019 09:07:00 GMT.  Free memory 289284
19:29:07.550 -> D: Display type is DUMMYTFT
19:29:07.550 -> D: Partition nvs found, 20480 bytes
19:29:07.550 -> D: Read 43 keys from NVS
19:29:07.550 -> D: pin_ir set to 35
19:29:07.550 -> D: pin_enc_clk set to 25
19:29:07.550 -> D: pin_enc_dt set to 26
19:29:07.550 -> D: pin_enc_sw set to 27
19:29:07.550 -> D: pin_tft_cs set to 15
19:29:07.550 -> D: pin_tft_dc set to 2
19:29:07.550 -> D: pin_tft_scl set to -1
19:29:07.550 -> D: pin_tft_sda set to -1
19:29:07.550 -> D: pin_tft_bl set to -1
19:29:07.550 -> D: pin_tft_blx set to -1
19:29:07.550 -> D: pin_sd_cs set to 21
19:29:07.550 -> D: pin_vs_cs set to 5
19:29:07.550 -> D: pin_vs_dcs set to 16
19:29:07.550 -> D: pin_vs_dreq set to 4
19:29:07.550 -> D: pin_shutdown set to -1
19:29:07.550 -> D: pin_shutdownx set to -1
19:29:07.550 -> D: pin_spi_sck set to 18
19:29:07.550 -> D: pin_spi_miso set to 19
19:29:07.550 -> D: pin_spi_mosi set to 23
19:29:07.550 -> D: GPIO0 is HIGH
19:29:07.928 -> D: GPIO2 is LOW, probably no PULL-UP
19:29:07.928 -> D: GPIO4 is HIGH
19:29:07.928 -> D: GPIO5 is HIGH
19:29:07.928 -> D: GPIO12 is HIGH
19:29:07.928 -> D: GPIO13 is HIGH
19:29:07.928 -> D: GPIO14 is HIGH
19:29:07.928 -> D: GPIO15 is HIGH
19:29:07.928 -> D: GPIO16 is HIGH
19:29:07.928 -> D: GPIO17 is HIGH
19:29:07.928 -> D: GPIO18 is HIGH
19:29:07.928 -> D: GPIO19 is HIGH
19:29:07.928 -> D: GPIO21 is HIGH
19:29:07.928 -> D: GPIO22 is HIGH
19:29:07.928 -> D: GPIO23 is HIGH
19:29:07.928 -> D: GPIO25 is LOW, probably no PULL-UP
19:29:07.928 -> D: GPIO26 is LOW, probably no PULL-UP
19:29:07.928 -> D: GPIO27 is HIGH
19:29:07.928 -> D: GPIO32 is HIGH
19:29:07.928 -> D: GPIO33 is HIGH
19:29:07.928 -> D: GPIO34 is LOW, probably no PULL-UP
19:29:07.928 -> D: GPIO35 is LOW, probably no PULL-UP
19:29:07.928 -> D: GPIO39 is LOW, probably no PULL-UP
19:29:07.928 -> D: gpio_00 will execute uppreset = 1
19:29:07.928 -> D: gpio_12 will execute upvolume = 2
19:29:07.928 -> D: gpio_13 will execute downvolume = 2
19:29:07.928 -> D: Enable pin 35 for IR
19:29:07.928 -> D: Start display
19:29:08.044 -> D: SD Card Mount Failed!
19:29:08.044 -> D: Create list with acceptable WiFi networks
19:29:08.044 -> D: Added NETGEAR35 to list of networks
19:29:08.044 -> D: Added SSID2 to list of networks
19:29:08.044 -> D: End adding networks
19:29:08.210 -> D: Scan Networks
19:29:10.337 -> D: Scan completed
19:29:10.337 -> D: Number of available networks: 7
19:29:10.337 -> D:  1 - Balticum 12-15            Signal: -73 dBm, Encryption WPA_WPA2_PSK, 
19:29:10.337 -> D:  2 - NETGEAR35                 Signal: -74 dBm, Encryption WPA2_PSK, Acceptable
19:29:10.337 -> D:  3 - Balticum12-2              Signal: -80 dBm, Encryption WPA_WPA2_PSK, 
19:29:10.372 -> D:  4 - PS4-B6645C2DBBC4          Signal: -84 dBm, Encryption WPA2_PSK, 
19:29:10.372 -> D:  5 - Rubuilijus                Signal: -84 dBm, Encryption WPA_WPA2_PSK, 
19:29:10.372 -> D:  6 - Telia-CA71C9-Greitas      Signal: -91 dBm, Encryption WPA2_PSK, 
19:29:10.372 -> D:  7 - Telia-BBF4B3-Greitas      Signal: -92 dBm, Encryption WPA2_PSK, 
19:29:10.372 -> D: End of list
[... 712 lines cut ...]
19:30:10.702 -> D: SPI semaphore not taken within 1000 ticks by CPU 0, id hspec
19:30:10.702 -> D: Semaphore is claimed by stopsong
19:30:11.599 -> D: Duration mp3loop 401
19:30:11.632 -> D: Song stopped incorrectly!
19:30:11.632 -> D: REG   Contents
19:30:11.632 -> D: ---   -----
19:30:11.632 -> D:   0 -   808
19:30:11.632 -> D:   1 -    33
19:30:11.665 -> D:   2 -     0
19:30:11.665 -> D:   3 -  6000
19:30:11.665 -> D:   4 -     0
19:30:11.665 -> D:   5 -  AC45
19:30:11.665 -> D:   6 -  1800
19:30:11.665 -> D:   7 -  1E06
19:30:11.698 -> D:   8 -     0
19:30:11.698 -> D:   9 -     0
19:30:11.698 -> D:   A -     0
19:30:11.698 -> D:   B -  4646
19:30:11.698 -> D:   C -     0
19:30:11.698 -> D:   D -     0
19:30:11.698 -> D:   E -     0
19:30:11.731 -> D:   F -     0
19:30:12.229 -> D: Metadata block 64 bytes
19:30:12.229 -> D: Streamtitle found, 50 bytes
19:30:12.229 -> D: StreamTitle='Jethro Tull - Bungle In The Jungle ';

Edzelf, 2019-12-16:
This time the wifi connection fails.  But we can see that the "stop song" function fails as well.  I will look into this tomorrow.

minzukas, 2019-12-16:
made new wiring, checked everything ten times. Now I also get VS1053 error in addition...
stop song however is also the same.

23:36:17.802 -> ⸮r⸮l⸮|⸮$⸮⸮⸮⸮⸮⸮$⸮⸮l2$⸮r⸮bb⸮l⸮⸮⸮⸮lllr⸮⸮⸮lp⸮⸮&'⸮⸮⸮#$⸮$⸮⸮⸮⸮l⸮⸮l`~⸮l⸮lrl⸮nn⸮⸮rnbb⸮⸮l⸮⸮⸮⸮⸮ ⸮';⸮⸮"⸮"#⸮⸮⸮"⸮##⸮⸮<⸮"#8⸮⸮⸮<⸮##'⸮⸮<⸮##⸮⸮⸮<⸮#$ ;⸮'⸮⸮⸮⸮⸮~⸮b⸮nnpn⸮⸮bll`p⸮n⸮br⸮⸮⸮⸮l#'⸮<"⸮$8⸮'⸮";⸮⸮⸮⸮$⸮⸮#'⸮<#$⸮⸮$8⸮'⸮"⸮⸮bn⸮|bl⸮⸮lp⸮n⸮#⸮⸮bn⸮|bl⸮l⸮⸮l⸮n⸮|⸮⸮⸮⸮⸮⸮ 
23:36:18.090 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 Dec 2019 09:07:00 GMT.  Free memory 294276
23:36:18.090 -> D: Display type is DUMMYTFT
23:36:18.090 -> D: Partition nvs found, 20480 bytes
23:36:18.090 -> D: Read 43 keys from NVS
23:36:18.090 -> D: pin_ir set to 35
23:36:18.090 -> D: pin_enc_clk set to 25
23:36:18.090 -> D: pin_enc_dt set to 26
23:36:18.090 -> D: pin_enc_sw set to 27
23:36:18.090 -> D: pin_tft_cs set to 15
23:36:18.090 -> D: pin_tft_dc set to 2
23:36:18.090 -> D: pin_tft_scl set to -1
23:36:18.090 -> D: pin_tft_sda set to -1
23:36:18.090 -> D: pin_tft_bl set to -1
23:36:18.090 -> D: pin_tft_blx set to -1
23:36:18.090 -> D: pin_sd_cs set to 21
23:36:18.090 -> D: pin_vs_cs set to 5
23:36:18.090 -> D: pin_vs_dcs set to 16
23:36:18.090 -> D: pin_vs_dreq set to 4
23:36:18.090 -> D: pin_shutdown set to -1
23:36:18.090 -> D: pin_shutdownx set to -1
23:36:18.090 -> D: pin_spi_sck set to 18
23:36:18.090 -> D: pin_spi_miso set to 19
23:36:18.090 -> D: pin_spi_mosi set to 23
23:36:18.090 -> D: GPIO0 is HIGH
23:36:18.090 -> D: GPIO2 is LOW, probably no PULL-UP
23:36:18.090 -> D: GPIO4 is HIGH
23:36:18.090 -> D: GPIO5 is HIGH
23:36:18.090 -> D: GPIO12 is HIGH
23:36:18.090 -> D: GPIO13 is HIGH
23:36:18.090 -> D: GPIO14 is HIGH
23:36:18.090 -> D: GPIO15 is HIGH
23:36:18.090 -> D: GPIO16 is HIGH
23:36:18.090 -> D: GPIO17 is HIGH
23:36:18.090 -> D: GPIO18 is HIGH
23:36:18.090 -> D: GPIO19 is HIGH
23:36:18.090 -> D: GPIO21 is HIGH
23:36:18.090 -> D: GPIO22 is HIGH
23:36:18.102 -> D: GPIO23 is HIGH
23:36:18.102 -> D: GPIO25 is HIGH
23:36:18.102 -> D: GPIO26 is HIGH
23:36:18.102 -> D: GPIO27 is HIGH
23:36:18.102 -> D: GPIO32 is HIGH
23:36:18.102 -> D: GPIO33 is HIGH
23:36:18.102 -> D: GPIO34 is LOW, probably no PULL-UP
23:36:18.102 -> D: GPIO35 is LOW, probably no PULL-UP
23:36:18.102 -> D: GPIO39 is LOW, probably no PULL-UP
23:36:18.102 -> D: gpio_00 will execute uppreset = 1
23:36:18.102 -> D: gpio_12 will execute upvolume = 2
23:36:18.102 -> D: gpio_13 will execute downvolume = 2
23:36:18.102 -> D: Enable pin 35 for IR
23:36:18.102 -> D: Start display
23:36:18.509 -> D: SD Card Mount Failed!
23:36:18.509 -> D: Create list with acceptable WiFi networks
23:36:18.509 -> D: Added NETGEAR35 to list of networks
23:36:18.509 -> D: Added SSID2 to list of networks
23:36:18.509 -> D: End adding networks
23:36:18.675 -> D: Scan Networks
23:36:20.811 -> D: Scan completed
23:36:20.811 -> D: Number of available networks: 7
23:36:20.811 -> D:  1 - NETGEAR35                 Signal: -77 dBm, Encryption WPA2_PSK, Acceptable
23:36:20.811 -> D:  2 - Balticum12-2              Signal: -79 dBm, Encryption WPA_WPA2_PSK, 
23:36:20.811 -> D:  3 - Rubuilijus                Signal: -81 dBm, Encryption WPA_WPA2_PSK, 
23:36:20.811 -> D:  4 - PS4-B6645C2DBBC4          Signal: -82 dBm, Encryption WPA2_PSK, 
23:36:20.811 -> D:  5 - Balticum 12-15            Signal: -84 dBm, Encryption WPA_WPA2_PSK, 
23:36:20.811 -> D:  6 - Telia-CA71C9-Greitas      Signal: -85 dBm, Encryption WPA2_PSK, 
23:36:20.934 -> D:  7 - Odeta                     Signal: -94 dBm, Encryption WPA2_PSK, 
[... 256 lines cut ...]
23:36:29.104 -> D: Semaphore is claimed by stopsong
23:36:29.137 -> D: SPI semaphore not taken within 1000 ticks by CPU 1, id hspec
23:36:29.137 -> D: Semaphore is claimed by stopsong
23:36:30.398 -> D: Song stopped incorrectly!
23:36:30.398 -> D: REG   Contents
23:36:30.398 -> D: ---   -----
23:36:30.398 -> D:   0 -   808
23:36:30.398 -> D:   1 -    33
23:36:30.398 -> D:   2 -     0
23:36:30.398 -> D:   3 -     0
23:36:30.431 -> D:   4 -     0
23:36:30.431 -> D:   5 -  1F40
23:36:30.431 -> D:   6 -     0
23:36:30.431 -> D:   7 -     0
23:36:30.431 -> D:   8 -     0
23:36:30.431 -> D:   9 -     0
23:36:30.464 -> D:   A -     0
23:36:30.464 -> D:   B -  4646
23:36:30.464 -> D:   C -     0
23:36:30.464 -> D:   D -     0
23:36:30.464 -> D:   E -     0
23:36:30.464 -> D:   F -     0
23:36:30.995 -> D: Metadata block 64 bytes
23:36:30.995 -> D: Streamtitle found, 60 bytes
23:36:30.995 -> D: StreamTitle='NPO Radio 1 - Langs de Lijn En Omstreken - EO';

Edzelf, 2019-12-16:
Looks like an SPI error.  Is the VCC (5 VOLT) on your VS1053 module strong enough?

minzukas, 2019-12-19:
Might be, i do not trust quality of jumper wires. However my esp32 was burned on other project so this one has to wait :)

Mohsin733, 2019-12-20:
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 770 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 780 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 790 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 800 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 810 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 820 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 830 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 840 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 850 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 860 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 870 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 880 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 890 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 900 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 910 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 920 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 930 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 940 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 950 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 960 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 970 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 980 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 990 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 1000 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    33
D:   2 -     0
D:   3 -  6000
D:   4 -    62
D:   5 -  AC45
D:   6 -  4051
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -  FFFF
D: Request 7 failed!
D: STOP requested
[... 108 lines cut ...]
D: SPI semaphore not taken within 640 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 650 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 660 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 670 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 680 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 690 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 700 ticks by CPU 1, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 710 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 720 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:128
D: Headerline: e not taken within 730 ticks by CPU 0, id hspectft

I also received the same .....

Edzelf, 2019-12-21:
The line:
`D: 1 - 33`
Reveals that you don't have a VS1053 , but an VS1003 chip mounted on your module instead.  The ESP32-Radio software will not work for this chip.

Tony763, 2019-12-30:
Hello, I have a same problem:
`D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 270 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   0 -   808

D:   1 -    33
D:   2 -  1A7B
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   3 -     0
D:   4 -     0
D: SPI semaphore not taken within 300 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   5 -  1F40
D:   6 -  2020
D: SPI semaphore not taken within 310 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   7 -     0

D:   8 -     0
D:   9 -     0
D: SPI semaphore not taken within 320 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   A -     0
D:   B -  F8F8
D: SPI semaphore not taken within 330 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   C -     0
D:   D -     0
D: SPI semaphore not taken within 340 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   E -     0

D:   F -     0
`

Odd is that previous version worked with it.

Edzelf, 2019-12-30:
Could you please check the printing on the chip?
It should say "VLSI Solution VS1053B".  For a different chip, the software must be modified.  I can not do that, because I only have VS1053 chips.

Tony763, 2019-12-30:
It is really VS1003 as identified, but still it worked. I can help with testing.

macikm, 2020-01-02:
"This is not a VS1053, but a VS1003 instead!"
same on my setup - but it works (if I comment out line "okay = false ;")

AST-esp-rasp, 2020-01-20:
Having the same issue as Minzukas
could it be because of the missing display because I use the dummy too?

Edzelf, 2020-01-20:
Missing display is allowed.  Do you have the right chip VS1053 on your board?

Derek-K, 2020-05-07:
While looking for differences between VS1053 and VS1053B, I come across [the datasheet](http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf) "12.1 Changes Between VS1033c and VS1053a/b Firmware" (page 87) not sure if that has any useful information.

Also on page 88 - "Known quirks, bugs, or features in VS1053b" and there is a '[patch](http://www.vlsi.fi/en/support/software/vs10xxpatches.html)' for VS1053B (last modified: 2020-04-29 V2.8)

About the VS1053 Patch can be found -> [here](http://www.vlsi.fi/fileadmin/software/VS10XX/vs1053b-patches.pdf)

Just my 2 cents.