Edzelf/ESP32-Radio #287: ESP32 as a bluetooth audio receiver

macikm, 2020-01-02:
When ESP32 has built-in BT module, is it possible to enhance this project of capability to read BT stream (eg. from phone) and play music from it?

macikm, 2020-10-07:
is it possible to make form ESP32-Radio a bluetooth speaker?
please, can You thing about it? :-)

Thanks a lot!

Edzelf, 2020-10-07:
Not in software.  You can add a bluetooth transmitter to the VS1053.
Something like [this](https://www.aliexpress.com/item/4000293080587.html?spm=a2g0o.productlist.0.0.1dbc3b40HvQ1Sb&algo_pvid=500f45cc-a395-4973-b21b-8deaaa6c8fe3&algo_expid=500f45cc-a395-4973-b21b-8deaaa6c8fe3-1&btsid=0b0a050b16020765967643496e6e9b&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_).

macikm, 2020-10-07:
> Not in software.  You can add a bluetooth transmitter to the VS1053.
> Something like this.

thanks for your reply
for now I have to connect directly to amplifier a BT receiver  or ESP radio - it cannot be connected in same time. So I'm thinking how to "integrate" function of BT receiver directly into ESP32 - without need of extra HW
(when a ESP32 has a capability of BT)