Edzelf/ESP32-Radio #289: VS1053 audio playback

nachette28, 2020-01-11:
Hi, I've just set up the ESP32 radio with the default parameters and followed the sugested pinout.
After correctly connecting to the WIFI network and detecting the vs1053, I try to play some of the multiple preset radio stations, but don't get any audio playback. Just the following error which I've not been able to track down.

D: Command client available
D: handlehttp started
D: Get command is: preset=10
D: Send reply for preset=10
D: Command: preset with parameter 10
D: STOP requested
D: Stopping client
D: New preset/file requested (10/0) from ihr/IHR_IEDM
D: Connect to new iHeartRadio host: IHR_IEDM
D: GET /api/livestream?version=1.5&mount=IHR_IEDMAAC&lang=en
D: Connected to playerservices.streamtheworld.com
D: Song stopped correctly after 10 msec
D: XML parser processing...
D: Bad xml status-code 404
D: Connect to new host 
D: Connect to  on port 80, extension /
D: Request  failed!

Here is the start-up log:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 21 Dec 2019 09:30:00 GMT.  Free memory 294100
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 45 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Start display
D: Create list with acceptable WiFi networks
D: Added Shitbox to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - ASUS_2.4                  Signal: -80 dBm, Encryption WPA2_PSK, 
D:  2 - Shitbox                   Signal: -84 dBm, Encryption WPA2_PSK, Acceptable
D: End of list
D: Command: ) with parameter ====" ;
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: gpio_14 with parameter stop
D: Command: gpio_17 with parameter resume
D: Command: gpio_34 with parameter station = icecast.omroep.nl:80/radio1-bb-mp3
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Try WiFi Shitbox
D: Connected to Shitbox
D: IP = 192.168.0.108
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Failed to obtain time!
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Song stopped correctly after 0 msec
D: Sync TOD
D: Request icecast.omroep.nl:80/radio1-bb-mp3 failed!
D: Duration mp3loop 1
D: Sync TOD, new value is 19:38:39

Any help would be greatly appreciated.
Thanks

Edzelf, 2020-01-11:
I see volume with parameter 0.  That will cause a nice silence.

nachette28, 2020-01-11:
Yes, I tried increasing and decreasing the volume, but from looking at the log, it seemed to me that there was a network related fault. is it normal for this error to pop up when changing radio stations:
D: Bad xml status-code 404
D: Connect to new host
D: Connect to on port 80, extension /
D: Request failed!

Edzelf, 2020-01-11:
Are you sure you can use GPIO 32 for DCS?  There may some limits for this pin.  The 404 code means that the requested URL does not exist, but at least icecast.omroep.nl:80/radio1-bb-mp3 should work.