Edzelf/ESP32-Radio #290: Missing .h files using Arduino IDE

rtgrandle, 2020-01-19:
I am new to building projects using the ESP processors. I am trying to build a simple project and an internet radio seemed to be easy enough. I have looked at several versions and am currently looking at your project. The hardware is simple enough. For the software I have downloaded  the Arduino IDE. Your instructions indicate that the software can be compiled on this platform. The first thing that happens is an error indicating that a header file NVS.h can not be found. I assume that this is because I am not including some library for the ESP. I seem to be encountering similar problems for any of the projects I try. Is there a common set of Libraries I should include and were are they listed in the directions?
Thanks

Edzelf, 2020-01-19:
Did you install the toolchain for the ESP32?  See https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/

rtgrandle, 2020-02-02:
I had not loaded the ESP3 toolchain. Newbee

closed 2020-02-02 by rtgrandle