Edzelf/ESP32-Radio #291: cannot compile: 'wifi_auth_mode_t' was not declared in this scope, bluetft.h:33:22: error: conflicting declaration 'Adafruit_ST7735* tft', some others

AloisKlingler, 2020-01-25:
Hi,

I am trying since yesterday to compile this source. It is failing and I do not know what is wrong. :-(
I am not a beginner neither with arduino nor with esp8266 / esp32.
In the meanwhile I have cleaned up all the arduino directories, including libraries, installed arduino fresh (latest 1.8.10) and:
 * added only the ESP32 boardURL (with ESP8266 URL there were always warnings about duplicate libraries, I saw the correct one in output used but as I were not sure ...) and installed it (version 1.0.4)
 * installed only pubsubclient (version 2.7.0) and Adafruit_ST7735 (version 1.5.10) libraries and its dependencies

It fails with 'wifi_auth_mode_t' was not declared in this scope first. ...\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp32\esp_wifi_types.h exists and contains the searched stuff.
As I saw the stuff only is used in some debugprints I commented-out these - but it is failing on other points and I do not know what to do for fixing the issue.

Anyone can help me?

Thanks!

BR
Alois

[compileoutput.txt](https://github.com/Edzelf/ESP32-Radio/files/4112189/compileoutput.txt)

Edzelf, 2020-01-25:
Looks like you have 2 types of tft displays configured.

AloisKlingler, 2020-01-25:
hello edzelf,

no, I have only one TFT active:
// Define (just one) type of display.  See documentation.
#define BLUETFT                        // Works also for RED TFT 128x160
//#define OLED                         // 64x128 I2C OLED
//#define DUMMYTFT                     // Dummy display
//#define LCD1602I2C                   // LCD 1602 display with I2C backpack
//#define ILI9341                      // ILI9341 240*320
//#define NEXTION                      // Nextion display. Uses UART 2 (pin 16 and 17)
//

I have exactly the sourcecode from repository downloaded and use the folder "ESP32_Radio" from the archive unmodiefied as sketch folder in arduino.

Thank you!
Best regards
Alois

Edzelf, 2020-01-25:
But dummytft.h is being compiled.
Try to remove references to this file and/or remove dummytft.*

AloisKlingler, 2020-01-26:
yeah, dummytft.h was this causing. but only removing away from the sketch-folder helps. even if I remove all references for dummytft out of the sketch it does not compile.

thank you for your help! :-)

AloisKlingler, 2020-02-21:
Ed, I was now struggling again with includes.
I am using windows 10, and at least there it is an issue if the name "Dummytft.h" starts with a capital D:

exit status 1
Dummytft.h: No such file or directory

As soon as I rename the file to dummytft.h and adapt the code accordingly it compiles correctly. Also after renaming it does not matter any more if there are other display definition header files on compiling.

I could not find any convention or a reason for this issue. but maybe you can reproduce.
If not maybe this post helps somebody with the same issue to save time. ;)

Best regards
Alois

Edzelf, 2020-02-24:
I also have Windows 10.  Dummytft.h starts with a capital "D".  The file Dummytft.h must be in the same directory as Esp32_radio.ino.  In your IDE there should be a tab with Dummytft.h.
The Esp32_radio.ino should contain these lines:
```
#ifdef DUMMYTFT
#include "Dummytft.h"                                    // For Dummy display
#endif

```