Edzelf/ESP32-Radio #292: IR Remote to Change Stations

clparrish, 2020-01-29:
Does a mod exist which allows an IR remote to change stations (PREV, NEXT)?

fletsche, 2020-01-29:
No need for a mod. You can just edit your config to do that. Here is an example for the Apple Remote:

`ir_601C = uppreset    # Apple Remote >>|
ir_901C = downpreset  # Apple Remote |<<`

clparrish, 2020-01-29:
Thanks for your reply. I tried this in the init file but it didn't work.

fletsche, 2020-01-30:
You also don't have to edit the init file. You can edit the preferences vie the web interface. The correct procedure is described in the documentation under "Preferences" and under "IR Interface". This also describes how you find the correct code for your remote.
I assume that with "init file" you mean "Esp32_radio_init.ino". Note that AFAIK this sketch is a separate tool for uploading preferences and has no unless you run this sketch separately from the main program.

clparrish, 2020-01-30:
fletsche,
Thank you for your comments and willingness to help. I have decided not to implement the remote feature, because my small remote does not have enough range to be practical.  I've tried other more robust remotes that I have, but none of them produce any codes at all in the serial monitor. Thank you again for your help.
Curt

closed 2020-01-30 by clparrish