Edzelf/ESP32-Radio #297: VS1053 is not working

Beppi4U, 2020-02-16:
I checked the wiring several times but there is no sound output from the VS1053 modul.
TFT on SPI workes 
Rotary Encoder workes 
SD card reader seems to work
WIFI workes and i can change radio stations as also the files on SD via web interface.....but without sound.
Can you please give me a hint on how to solve the problem?
As far as i understand the Serial output its a SPI problem!?
Andreas  

SERIAL OUTPUT------  
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 21 Dec 2019 09:30:00 GMT.  Free memory 294100
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Locate mp3 files on SD, may take a while...
D: 18 tracks on SD
D: Create list with acceptable WiFi networks
D: Added WLAN-DLRPMN to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 6
D:  1 - Telekom_FON               Signal: -55 dBm, Encryption OPEN, 
D:  2 - WLAN-DLRPMN               Signal: -56 dBm, Encryption WPA2_PSK, Acceptable
D:  3 - EasyBox-241519            Signal: -60 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - LoededDiper               Signal: -90 dBm, Encryption WPA2_PSK, 
D:  5 - WLAN-ZMSGZX               Signal: -93 dBm, Encryption WPA2_PSK, 
D:  6 - WLAN-316264               Signal: -95 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 22
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 SPI error. SB:0000 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:012C R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0258 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0384 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:04B0 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:05DC R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0708 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0834 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0960 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0A8C R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0BB8 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0CE4 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0E10 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:0F3C R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:1068 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:1194 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:12C0 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:13EC R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:1518 R1:FFFF R2:FFFF
D: VS1053 SPI error. SB:1644 R1:FFFF R2:FFFF
D: This is not a VS1053, but a VS1103 instead!
D: Connect to WiFi
D: Connected to WLAN-DLRPMN
D: IP = 192.168.2.111
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 20:31:09
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: SPI semaphore not taken within 120 ticks by CPU 1, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 130 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 140 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 150 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: Connected to server
D: Duration mp3loop 1
D: SPI semaphore not taken within 160 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 170 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 180 ticks by CPU 1, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 190 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 200 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 210 ticks by CPU 1, id hspec
D: Semaphore is claimed by stopsong
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-a taken within 220 ticks by CPU 0, id hspec

SERIAL OUTPUT END-------

Beppi4U, 2020-02-16:
…...hell!  As i put out the SD Card Reader modul it started to work!
Have to check the SD Card modul wiring…….
Is it possible to use the TFT OnBoard Card reader for the mp3 files?

Beppi4U, 2020-02-17:
every time the same problem......Read the F%§&ing manual! 
Seems that i totaly misunderstood the wiring plan. of Course i can use the 1.8TFT SD Card Slot for playing MP3's. Just soldered the Pins wired it corectly and.....tada! It workes without any problem!
Sorry for being a DAU (Dümmster anzunehmender User) ......case closed! ;-)
BTW: Big thanks for your effort. It is a great Project!

closed 2020-02-17 by Beppi4U

t-c-o-d-e, 2020-03-05:
> I checked the wiring several times but there is no sound output from the VS1053 modul.
> TFT on SPI workes
> Rotary Encoder workes
> SD card reader seems to work
> WIFI workes and i can change radio [...]

Can you please explain what exactly you do wrong first time? I'm having same issues with SPI as you had. Thank you!