Edzelf/ESP32-Radio #298: High active Sensor how to / STOP after time

p1mike, 2020-02-17:
Hi Ed, my idea is, to use a motion control sensor to switch the radio on for 10-15minutes or better to switch between STOP and PLAY mode. But my motion sensor is HIGH active and I'm not sure how to change your default procedures for LOW active GPIOs. So I wish a suggestion for this and if you have: best point in code to change the radio mode after time and also for change to play mode.
Sending ESP32 into deep sleep is (for now) not an option, because boot time is too long (15s). 
Best 
Mike

Edzelf, 2020-02-18:
Look for the function "scandigital".  It should be near line 3096.  Change the part in line 3118-3130:

p1mike, 2020-02-20:
Hi Ed,
thank you. I have changed pin initialising in setup and it works:

  for ( i = 0 ; (pinnr = progpin[i].gpio) >= 0 ; i++ )   // Check programmable input pins
  {
    if (pinnr == motion_pin)
      pinMode ( pinnr, INPUT ) ;                    // Input for motion control sensor
    else
      pinMode ( pinnr, INPUT_PULLUP ) ;                    // Input for control button
    delay ( 10 ) ;