Edzelf/ESP32-Radio #300: Scroll speed LCD 1602

Thoky81, 2020-02-21:
It is possible to change the scroll speed on 1602 lcd display? Thanks

buntings, 2022-01-23:
I have the same question, would like to increase the scroll speed.

Edzelf, 2022-01-23:
That would not be so hard to find.....

buntings, 2022-01-24:
I would imagine it is easy to find if you know where it is! I looked, then I searched the issues, looked again and only then asked the question......I have read your reply, looked again.....and I still cannot find the parameter. Sorry.

Edzelf, 2022-01-24:
Every second, the time will be displayed on the display.  On a 1602, there will be a scroll of the text if necessary.  So you can display the time every half second.  Insert the lines:
```
#ifdef LCD1602I2C
  if ( ( count10sec % 10 ) == 5 )                 // Half second over?
  {
    time_req = true ;                             // Yes, show current time request
  }
#endif

```
Just before:

`  if ( ( count10sec % 10 ) == 0 )                 // One second over?
`

In the "timer100()" function.

```