Edzelf/ESP32-Radio #302: Status refresh

Beppi4U, 2020-03-02:
How can i refresh the Status HttpGet every 20 seconds?
…..like     <meta http-equiv="refresh" content="20">    refreshes the whole Web site.
My Target is to have a Fresh Status of(Mp3) title when the Playlist goes on with a new title.

Edzelf, 2020-03-03:
You may add a line like:
`<meta http-equiv="refresh" content="20">`
Just before line 10 of index_html.h.
Note that this may effect the audio stream.

Beppi4U, 2020-03-03:
Already tried that out! It's not working! My understanding of the HTLM code is that the Button event HttpGet"Status" command forces the HMTL handler to update the status line and provide the Mp3 Tags or whatsoever!
Either the code should send a status refresh or the HTML code should provide the function! (Sending the GET command every X sec.).
I check if the HTML is able to do so......i will come back later.