Edzelf/ESP32-Radio #306: Several m3us on SD card don't work

egrue, 2020-03-21:
In general it is great work! Internet radio was instantly working.
[image]

But I want to have 4 GPIOs configured as internet radio streams – this works fine
and 4 additional GPIOs configured to play back lists of mp3 files (and cycle inside this list with NEXT).

On the SD card I made 9 folders and copied a list of mp3 files into each folder.
Additionally I generated an m3u file for each list and added it to the mp3 files in the folder.
Here is a screen print of the SD card, e.g. in folder 3 you see the 30 tracks (mp3 and Herbst.m3u)
In total these are 258 mp3 files in 9 folders and 9 m3u files, 2,88GB.

(See attached image) 

In the radio’s config I wrote:

#
gpio_02 = uppreset = 1
gpio_13 = station = 92.255.229.74:8000/kids-hits-junior   #  RUS kids-hits-junior
gpio_14 = station = 167.114.246.177:8119/stream   #Pucki Geschichten
gpio_15 = resume
gpio_17 = downpreset = 1
gpio_22 = stop
gpio_32 = station = wdr-edge-101d-dus-dtag-cdn.cast.addradio.de/wdr/diemaus/live/mp3/high # Die Maus live
gpio_33 = station = stream.laut.fm/kinderlieder-123 # Kinderlieder 123
gpio_34 = station = localhost/8/RusMaerchen.m3u
gpio_35 = station = localhost/9/Klassik.m3u
gpio_36 = station = localhost/1/Fruehling.m3u
gpio_39 = station = localhost/7/Maerchen.m3u
#

The strange thing I experience is that all buttons gpio_34 .. 39 play localhost/1/Fruehling.m3u.  maybe, because it is the first one.

The log stream in the serial monitor shows (if button of GPIO_39 is pressed):

22:04:18.040 -> D: GPIO_39 is now LOW, execute station = localhost/7/Maerchen.m3u
22:04:18.075 -> D: Command: station with parameter localhost/7/Maerchen.m3u
22:04:18.075 -> D: Playing localhost/7/Maerchen.m3u 
This means: The configuration is read correctly , in this case GPIO_39, and it should play localhost/7/Maerchen.m3u. But from the speakers comes localhost/1/Fruehling.m3u

It seems, that the software reads the first m3u, and then keeps it in mind, and plays it again and again, even if another m3u is orderd to play.

Did you experience this before?

How should I arrange the files and configure the GPIOs that I get what I need?

Which additional test can I do?

I tried it without m3us in several folders: dosn’t work
I tried it with m3us and mp3s in the root directory of the SD card: doesn’t work, same error.

I am using the version "Sat, 21 Dec 2019 09:30:00 GMT".

egrue, 2020-03-21:
Additionally I formatted another SD card, 2GB, with FAT, full format,  (the first one was FAT32) and then copied 
1. The directory 7 containing Maerchen.m3u and the corresponding mp3s
2. The directory 1  containing Fruehling.m3u and the corresponding mp3s
to it.

The config:
…
gpio_33 = station = stream.laut.fm/kinderlieder-123 # Kinderlieder 123
gpio_34 = station = localhost/7/Maerchen.m3u 
gpio_39 = station = localhost/1/Fruehling.m3u 
#

When I press the button of GPIO_34 Maerchen.m3u is played.
When I press the button of GPIO_39 Maerchen.m3u is played, too.

--------------------------------

Then another test:

I formatted the SD card, 2GB, with FAT and then copied 
1. The directory 7 containing Fruehling.m3u and the corresponding mp3s
2. The directory 1  containing Maerchen.m3u and the corresponding mp3s
to it (i.e. I switched the filenames).

The config:
…
gpio_34 = station = localhost/1/Maerchen.m3u
gpio_39 = station = localhost/7/Fruehling.m3u
#

When I press the button of GPIO_34 Fruehling.m3u is played.
When I press the button of GPIO_39 Fruehling.m3u is played, too.

------

In summary it looks like that always the m3u is played that is in the physically first directory on the SD.

Edzelf, 2020-03-21:
Can you give an example of the first few linws of a .m3u file.

egrue, 2020-03-21:
Sure. I made the m3us with the latest version of VLC on Win10.

Fruehling.m3u:

#EXTM3U
#EXTINF:120,01-Frühling.mp3
01-Fr%C3%BChling.mp3
#EXTINF:177,02-Ich lieb den Frühling.mp3
02-Ich%20lieb%20den%20Fr%C3%BChling.mp3
#EXTINF:207,03- Stups, der kleine Osterhase.mp3
03-%20Stups%2C%20der%20kleine%20Osterhase.mp3
#EXTINF:219,04-Hier ein Ei und dort ein Ei.mp3
04-Hier%20ein%20Ei%20und%20dort%20ein%20Ei.mp3
#EXTINF:338,05-Simone Sommerland, Karsten Gl05-Simone%20Sommerland%2C%20Karsten%20Gl%EF%BF%BCck%20und%20die%20Kita-Fr%EF%BF%B6sche%20_%20Die%20Vogelhochzeit.mp3
#EXTINF:288,06-Simone Sommerland, Karsten Gl06-Simone%20Sommerland%2C%20Karsten%20Gl%EF%BF%BCck%20und%20die%20Kita-Fr%EF%BF%B6sche%20_%20Hoppelhase%20Hans.mp3
#EXTINF:87,07-Simone Sommerland, Karsten Gl07-Simone%20Sommerland%2C%20Karsten%20Gl%EF%BF%BCck%20und%20die%20Kita-Fr%EF%BF%B6sche%20_%20Jetzt%20f%EF%BF%A4ngt%20das%20sch%EF%BF%B6ne%20Fr%EF%BF%BChjahr%20an.mp3
…

Maerchen.m3u:

#EXTM3U
#EXTINF:559,Julia Preuß - Der Marionettenspieler
#EXTVLCOPT:file-caching=300
02da30528e_Andersen_Der%20Marionettenspieler.mp3
#EXTINF:866,Philipp Oehme - Der Traum der alten Eiche
#EXTVLCOPT:file-caching=300
09dff2bb35_Andersen_Der_Traum_der_alten_Eiche.mp3
#EXTINF:567,Rainer Koschorz, Wolfgang Gerber – Dornröschen
…

Edzelf, 2020-03-21:
Okay.  I will test it later.  But can you try the same setup with filenames that do not contain spaces and special characters?
And use the full path of the songs in the playlist, like:
localhost/8/song1.mp3

egrue, 2020-03-22:
Now I made „nice“ m3u s:
As an intermediate step with lines like  7/song3.mp3

#EXTM3U
#EXTINF:124,01Fruehling.mp3
7/01Fruehling.mp3
#EXTINF:177,02Fruehling.mp3
7/02Fruehling.mp3
#EXTINF:207,03Fruehling.mp3
7/03Fruehling.mp3
#EXTINF:219,04Fruehling.mp3
7/04Fruehling.mp3
#EXTINF:338,05Fruehling.mp3
7/05Fruehling.mp3
#EXTINF:288,06Fruehling.mp3
7/06Fruehling.mp3
#EXTINF:87,07Fruehling.mp3
7/07Fruehling.mp3
#EXTINF:120,08Fruehling.mp3
7/08Fruehling.mp3

#EXTM3U
#EXTINF:419,Thomas Dehler - Die Wichtelmänner
1/Maerchen1.mp3
#EXTINF:559,Julia Preuß - Der Marionettenspieler
1/Maerchen2.mp3
#EXTINF:1217,Julia Preuß - Das hässliche junge Entlein
1/Maerchen3.mp3
#EXTINF:420,Rainer Koschorz, Wolfgang Gerber - Rumpelstilzchen
1/Maerchen4.mp3
#EXTINF:2943,Lydia Herms - Die zwei Brüder
1/Maerchen5.mp3
#EXTINF:1556,Wilhelm Hauff - Die Geschichte von dem Gespensterschiff
1/Maerchen6.mp3
#EXTINF:2360,Brigitte Trübenbach - Das goldene Königreich
1/Maerchen7.mp3

After full format I copied
1.	Maerchen.m3u to the root dir
2.	Fruehling.m3u to the root dir
3.	The whole dir 1 (containing the Maerchen mp3s)
4.	The whole dir 7 (containing the Fruehling mp3s)
Config:
gpio_34 = station = localhost/Fruehling.m3u
gpio_39 = station = localhost/Maerchen.m3u
#
I pressed GPIO_34 first, then GPIO_39. Both play Maerchen.m3u.

-------------

After format (a quick format is also ok for these tests) I copied
1.	Maerchen.m3u to the root dir
2.	Fruehling.m3u to the root dir
3.	The whole dir 7 (containing the Fruehling mp3s)
4.	The whole dir 1 (containing the Maerchen mp3s)

Config:
gpio_34 = station = localhost/Fruehling.m3u
gpio_39 = station = localhost/Maerchen.m3u
#
I pressed GPIO_39 first, then GPIO_34. Both play Frueling.m3u.

-------------

So in summary it looks like that the software knows which m3u to play on which button.
But the physically first directory of mp3s is played.

BUT: Then I discovered something more strange: When I press GPIO_34 and GPIO_39 many times, sometimes the playback changes to the other m3u/directory, then, if constantly pressing buttons (always with a pause of some seconds), both buttons play the other m3u/directory. When I go on pressing buttons, the whole playback changes back to the first m3u after some time, and so on.

------------
[... 77 lines cut ...]
10:05:59.445 -> D: ID3 TYER = 2004
10:05:59.445 -> D: ID3 TCON = (101)
10:05:59.445 -> D: ID3 TPE1 = Wilhelm Hauff
 …
10:06:03.691 -> D: GPIO_39 is now LOW, execute station = localhost/Maerchen.m3u
10:06:03.725 -> D: Command: station with parameter localhost/Maerchen.m3u
10:06:03.725 -> D: Playing localhost/Maerchen.m3u
10:06:03.725 -> D: STOP requested
10:06:03.725 -> D: SD_currentnode is 4,6,0,0, curnod is 4,6,0,0, delta is 1
10:06:03.725 -> D: getSDfilename requested node ID is 4,7,0,0
10:06:04.029 -> D: Found ID3 info
 …
10:06:07.293 -> D: GPIO_39 is now LOW, execute station = localhost/Maerchen.m3u
10:06:07.293 -> D: Command: station with parameter localhost/Maerchen.m3u
10:06:07.293 -> D: Playing localhost/Maerchen.m3u
10:06:07.293 -> D: STOP requested
10:06:07.361 -> D: SD_currentnode is 4,7,0,0, curnod is 4,7,0,0, delta is 1
10:06:07.361 -> D: getSDfilename requested node ID is 5,1,0,0     <------------here it starts to play the second mp3 directory
10:06:07.463 -> D: Found ID3 info 

In the Maerchen directory are 7 mp3 files. After 4,7,0,0 it changes to 5,1,0,0.

Here I think I understood what is going on:
The software reads the m3u files, and reads the config properly. It „knows“ which file is to be played next as specified in the m3u.
But on the lower level the software simply plays the mp3 files as they are physically located on the SD: First the 7 mp3 files of the (first) Maerchen directory, and then the 8 mp3 files of the (second) Fruehling directory.

Edzelf, 2020-03-23:
Playlists on SD card were not supported.  The new version will support it.  Maybe you can test it.  An example of my playlist is:
```
#EXTM3U
#EXTINF:124,01bachfugue.mp3
MAP0/bachfugue.mp3
#EXTINF:177,02gtr-nylon22.mp3
MAP0/gtr-nylon22.mp3
#EXTINF:207,03harpsi-cs.mp3
MAP0/harpsi-cs.mp3
#EXTINF:100,01viola2.mp3
MAP4/viola2.mp3
#EXTINF:101,02violin.mp3
MAP4/violin.mp3
#EXTINF:102.03vln-lin-cs.mp3
MAP4/vln-lin-cs.mp3

```
MAP0..MAP4 are subdirectories on the SD card.  No need to prepend "localhost/" to the entries.

egrue, 2020-03-23:
Hey, ED you hacked it in this short time? Cool!
Of course you know your code, then it's somewhat easier, but anyway...

OK, you put your playlist into the root directory, right?

And: Is it also solved that all buttons/GPIOs play the same list of files?

Edzelf, 2020-03-23:
I think the playlist can be in any directory.  Plaese tset it.

egrue, 2020-03-23:
OK

egrue, 2020-03-23:
Works, with 2 M3Us and 2 buttons!  :)
But if I press one button again, the playlist starts again.
In the old version again pressing a button alsways startet the next file.
But we have the NEXT and PREV commands, I will test them now...

egrue, 2020-03-24:
I thing if in "playlist mode" the "uppreset" and "downpreset" commands should work on the current playlist and cycle through it, i.e. make a "wrap around" in one or the other direction.
Currentlich they work on the bare mp3s, which gives more or less random playback.
 at the end or beginning.
And I think, when all files of the playlist have been played, it should start with Nr. 1 again, and not switch to internet radio.

Then the playlist szenario would be complete.

Edzelf, 2020-03-25:
uppreset and downpreset are working now.  End of playlist will restart the playlist.

egrue, 2020-03-25:
Ed, you are the best one!

egrue, 2020-03-25:
... a little thing:
When I start the radio, internetradio starts. With uppreset and downpreset I can change channels.
When I switch to a playlist, with uppreset and downpreset I can change the songs.
(In both cases it doesn't cycle, this is ok and consistent.)
But when after listening to a playlist I change back to internet radio, uppreset and downpreset don't work anymore.

Edzelf, 2020-03-26:
Sorry, I cannot reproduce this bug.

egrue, 2020-03-26:
I reproduced it 3 times, each with your latest version, each with fresh reboot.
I can try to make a step-by-step list today.

egrue, 2020-03-26:
Yesterday I found: when you change form Internet radio to playlist, this is not enough to disturb internet radio up/down. You need to use up/down in the playlist, then change back ti internet radio. Then up/down doesn't work.

I made a few small changes to your code, e.g. commenting out ADC for one more GPIO etc. Now I will use your fresh source, maybe a smaller config.

I guess that in the context switch from playlist back to internet radio some variable is not set back properly. We will see...

egrue, 2020-03-26:
Now I used your fresh sources, exchanged  #define BLUETFT for DUMMYTFT, compiled and loaded it to the ESP32.
The radio startet,
then jumped to internet radio preset0 (wdr..),
then I activated GPIO39, so it switched to Maerchen.m3u and started the first mp3 (3federn.mp3).
then I activated GPGIO17 (UPPRESET), so it switched to the second entry,
then I activated GPGIO17 (UPPRESET) again, so it switched to the third entry

Then I activated GPGIO33, so it switched to internet radio again (Kinderlieder 123, 2nd entry),
then I activated GPGIO17 (UPPRESET), it is recognized, but the radio plays the same entry again,
then I activated GPGIO17 (UPPRESET) again, it is recognized, but the radio plays the same entry again. ie the radio stations are no longer changed.

In this szenario I only used UPPRESET, you should also check DOWNPRESET.

Here is the corresponding log:
...
16:57:45.761 -> D: Start server for commands
16:57:45.761 -> D: Network found. Starting mqtt and OTA
16:57:45.761 -> D: MDNS responder started
16:57:46.780 -> D: Rotary encoder is enabled
16:57:46.780 -> D: STOP requested
16:57:46.780 -> D: New preset/file requested (0/0) from wdr-edge-101d-dus-dtag-cdn.cast.addradio.de/wdr/diemaus/live/mp3/high
16:57:46.780 -> D: Connect to new host wdr-edge-101d-dus-dtag-cdn.cast.addradio.de/wdr/diemaus/live/mp3/high
16:57:46.780 -> D: Connect to wdr-edge-101d-dus-dtag-cdn.cast.addradio.de on port 80, extension /wdr/diemaus/live/mp3/high
16:57:46.780 -> D: Song stopped correctly after 0 msec
16:57:46.953 -> D: Connected to server
16:57:46.953 -> D: Duration mp3loop 1
16:57:47.124 -> D: Switch to HEADER
16:57:47.124 -> D: Headerline: Content-Type: audio/mpeg
16:57:47.124 -> D: audio/mpeg seen.
16:57:47.124 -> D: Headerline: icy-description:http://www.wdr.de/radio/radiotext/streamtitle_kiraka.txt
16:57:47.124 -> D: Headerline: icy-genre:wdr
16:57:47.157 -> D: Headerline: icy-name:Die Maus, Westdeutscher Rundfunk Koeln
16:57:47.157 -> D: Headerline: icy-pub:1
16:57:47.157 -> D: Headerline: icy-url:http://www.wdr.de
16:57:47.157 -> D: Headerline: Server: dg-wdr-https_dus-dtag_edge_d30050acea897b9247cfd33952665b3f
16:57:47.157 -> D: Headerline: Cache-Control: no-cache, no-store
16:57:47.157 -> D: Headerline: Access-Control-Allow-Origin: *
16:57:47.157 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
16:57:47.191 -> D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
16:57:47.191 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
16:57:47.191 -> D: Headerline: Connection: Close
16:57:47.191 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
16:57:47.191 -> D: Headerline: icy-metaint:16000
16:57:47.191 -> D: Switch to DATA, bitrate is 0, metaint is 16000
16:57:47.191 -> D: Duration mp3loop 71
16:57:47.394 -> D: Metadata block 80 bytes
16:57:47.394 -> D: Streamtitle found, 77 bytes
16:57:47.394 -> D: StreamTitle='Die Sendung mit der Maus - Lach- und Sachgeschichten zum H⸮ren';
16:58:03.493 -> D: Rotation count 1
16:58:03.661 -> D: Rotation count 1
16:58:27.431 -> D: GPIO_39 is now LOW, execute station = localhost/Maerchen.m3u
16:58:27.431 -> D: Command: station with parameter localhost/Maerchen.m3u
16:58:27.431 -> D: Playing localhost/Maerchen.m3u
16:58:27.431 -> D: STOP requested
16:58:27.431 -> D: Stopping client
16:58:27.941 -> D: Playlist request, entry 1
16:58:27.975 -> D: Read from playlist
16:58:27.975 -> D: Playlistdata: #EXTM3U
16:58:27.975 -> D: Playlistdata: #EXTINF:410,Johannes Ackner - Die drei Federn
16:58:27.975 -> D: Playlistdata: 7/3federn.mp3
16:58:28.009 -> D: Entry 1 in playlist found: 7/3federn.mp3
16:58:28.009 -> D: Closed SD file
16:58:28.043 -> D: Found ID3 info
16:58:28.043 -> D: ID3 TYER = 2012
16:58:28.077 -> D: Duration mp3loop 148
16:58:28.383 -> D: Song stopped correctly after 20 msec
16:58:28.383 -> D: Duration mp3loop 155
17:00:29.152 -> D: GPIO_17 is now LOW, execute uppreset = 1
17:00:29.152 -> D: Command: uppreset with parameter 1
17:00:29.185 -> D: Command accepted
[... 180 lines cut ...]
17:02:27.601 -> D: Switch to HEADER
17:02:27.601 -> D: Headerline: Content-Type: audio/mpeg
17:02:27.601 -> D: audio/mpeg seen.
17:02:27.601 -> D: Headerline: icy-description:Kinderlieder fr die Allerkleinsten
17:02:27.601 -> D: Headerline: icy-genre:Kinderlieder
17:02:27.601 -> D: Headerline: icy-name:kinderlieder-123
17:02:27.634 -> D: Headerline: icy-pub:1
17:02:27.634 -> D: Headerline: icy-url:laut.fm/kinderlieder-123
17:02:27.634 -> D: Headerline: Server: Icecast 2.4.0-kh8
17:02:27.634 -> D: Headerline: Cache-Control: no-cache, no-store
17:02:27.634 -> D: Headerline: Access-Control-Allow-Origin: *
17:02:27.634 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
17:02:27.634 -> D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
17:02:27.669 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
17:02:27.669 -> D: Headerline: Connection: Close
17:02:27.669 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
17:02:27.669 -> D: Headerline: icy-metaint:16000
17:02:27.669 -> D: Switch to DATA, bitrate is 0, metaint is 16000
17:02:27.939 -> D: Song stopped correctly after 20 msec
17:02:28.449 -> D: Metadata block 48 bytes
17:02:28.449 -> D: Streamtitle found, 44 bytes
17:02:28.449 -> D: StreamTitle='Robert Metcalf - Hallo, Hallo';
17:02:31.438 -> D: Metadata block 48 bytes
17:02:31.438 -> D: Streamtitle found, 39 bytes
17:02:31.438 -> D: StreamTitle='Saitentwist - Lokomotive';

Edzelf, 2020-03-27:
I still cannot reproduce the bug.  What I see in the logging is the sentence "D: Preset is now **1**".  When I try to do the same I see the preset changing to 2, 3, 4, ...
Could you try to use the webinterface instead of the buttons?

egrue, 2020-03-27:
Hm. Maybe you can introduce some additional clever debug outputs in the code where the switch of the radio station is done? Unfortunately I cannot debug with breakpoints here.
In my log you see that the software knows that it shall switch the station, but it calls preset1 all the time.

Some additional info: In the past (before the introduction of playlists on SD) I sometimes saw, that UP/DOWN of radio stations did not work properly: Mostly it works, but sometimes it plays another station, especially at or near the 0-/1-entry (first/second preset).
Can the special entries in my config (with comments) be the reason?
Or can it be a problem in NVRAM?
But I use a new ESP32-WROOM-32D, it should be ok.

Now I also tested it as you suggested:
The radio startet,
then jumped to internet radio preset0 (wdr..),
then I activated GPIO39, so it switched to Maerchen.m3u and started the first mp3 (3federn.mp3).
then I activated NEXT via the Control http page, so it switched to the second entry,
then I activated NEXT via the Control http page again, so it switched to the third entry
Then I activated GPGIO33, so it switched to internet radio again (Kinderlieder 123, 2nd entry),
then I activated NEXT via the Control http page, it is recognized, but the radio plays the same entry again,
then I activated NEXT via the Control http page again, it is recognized, but the radio plays the same entry again. ie the radio stations are no longer changed.

In this szenario I only used NEXT (UPPRESET). 
It is exactely the same result as activation GPIO17.

I sent you my Config by email. Maybe you could test with my config?

Here ist he corresponding log:
…
13:08:26.532 -> D: Rotation count 1
13:08:26.601 -> D: Rotation count 1
13:08:36.839 -> D: GPIO_39 is now LOW, execute station = localhost/Maerchen.m3u
13:08:36.839 -> D: Command: station with parameter localhost/Maerchen.m3u
13:08:36.839 -> D: Playing localhost/Maerchen.m3u
13:08:36.839 -> D: STOP requested
13:08:36.839 -> D: Stopping client
13:08:37.317 -> D: Playlist request, entry 1
13:08:37.384 -> D: Read from playlist
13:08:37.384 -> D: Playlistdata: #EXTM3U

13:08:37.384 -> D: Playlistdata: #EXTINF:410,Johannes Ackner - Die drei Federn

13:08:37.384 -> D: Playlistdata: 7/3federn.mp3

13:08:37.418 -> D: Entry 1 in playlist found: 7/3federn.mp3

13:08:37.418 -> D: Closed SD file
13:08:37.452 -> D: Found ID3 info
13:08:37.452 -> D: ID3 TYER = 2012
13:08:37.792 -> D: Song stopped correctly after 20 msec
13:09:08.357 -> D: Command client available
13:09:08.357 -> D: handlehttp started
13:09:08.357 -> D: Get command is: uppreset=1
13:09:08.391 -> D: Send reply for uppreset=1
13:09:08.391 -> D: Command: uppreset with parameter 1
13:09:08.391 -> D: STOP requested
13:09:08.391 -> D: Playlist request, entry 2
13:09:08.391 -> D: Read from playlist
13:09:08.391 -> D: Playlistdata: #EXTM3U

13:09:08.391 -> D: Playlistdata: #EXTINF:410,Johannes Ackner - Die drei Federn

13:09:08.391 -> D: Playlistdata: 7/3federn.mp3

13:09:08.424 -> D: Entry 1 in playlist found: 7/3federn.mp3

13:09:08.424 -> D: Playlistdata: #EXTINF:825,Thomas Dehler - Allerleirauh

13:09:08.424 -> D: Playlistdata: 7/Allerleirauh.mp3

13:09:08.424 -> D: Entry 2 in playlist found: 7/Allerleirauh.mp3

13:09:08.526 -> D: Closed SD file
13:09:08.560 -> D: Found ID3 info
[... 144 lines cut ...]
13:13:14.757 -> D: Headerline: icy-metaint:16000
13:13:14.757 -> D: Switch to DATA, bitrate is 0, metaint is 16000
13:13:14.961 -> D: Song stopped correctly after 20 msec
13:13:15.473 -> D: Metadata block 48 bytes
13:13:15.473 -> D: Streamtitle found, 39 bytes
13:13:15.473 -> D: StreamTitle='Tom Lugo - Blaulichtlied';
13:13:31.367 -> D: Metadata block 48 bytes
13:13:31.367 -> D: Streamtitle found, 47 bytes
13:13:31.367 -> D: StreamTitle='Kati & Lars - Ich bin ein Frosch';
13:14:55.345 -> D: Metadata block 64 bytes
13:14:55.345 -> D: Streamtitle found, 53 bytes
13:14:55.345 -> D: StreamTitle='Christiane Weber - Pillipallipullipups';
13:16:21.362 -> D: Metadata block 48 bytes
13:16:21.362 -> D: Streamtitle found, 40 bytes
13:16:21.362 -> D: StreamTitle='Mathi - Miese fiese Riese';
13:19:54.315 -> D: Metadata block 48 bytes
13:19:54.350 -> D: Streamtitle found, 46 bytes
13:19:54.350 -> D: StreamTitle='START_AD_BREAK - START_AD_BREAK';
13:19:55.336 -> D: Metadata block 64 bytes
13:19:55.336 -> D: Streamtitle found, 61 bytes
13:19:55.336 -> D: StreamTitle='Detlev Jöcker - Wann Fängt Der Frühling An?';
13:22:56.082 -> D: Metadata block 64 bytes
13:22:56.082 -> D: Streamtitle found, 56 bytes
13:22:56.082 -> D: StreamTitle='Ulf der Spielmann - Der kleine Kitzelmann';
[image]

egrue, 2020-03-28:
I confirm that it works with a configuration like:

gpio_13 = preset = 7   #  comment

if the preset is already in the list.