Edzelf/ESP32-Radio #309: When mounting SD card (problem)

Jefke1, 2020-03-31:
Hi, radio plays very well, when i connect the sd card, i no longer get any sound from the radio and from the sd card . What am I doing wrong ??
Chip = VLSI 1053b

Log before:
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0

Afther: With SD card

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 294052
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 0
D: Pin 0 is already reserved!
D: pin_enc_clk set to 0
D: Pin 0 is already reserved!
D: pin_enc_dt set to 0
D: Pin 0 is already reserved!
D: pin_enc_sw set to 0
D: Pin 0 is already reserved!
D: pin_tft_cs set to 0
D: Pin 0 is already reserved!
D: pin_tft_dc set to 0
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 0 for IR
D: Start display
D: Init Nextion, pins 16,17
D: Nextion command cls BLACK
D: Nextion command page 0
D: Locate mp3 files on SD, may take a while...
D: SD directory is /
[... 144 lines cut ...]
D:   F -  FFFF
D: Nextion t0.txt="ESP32-Radio"
D: Switch to HEADER
D: Duration mp3loop 3
D: Headerline: icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k
D: Headerline: icy-genre:Love and Romance
D: Headerline: icy-url:http://www.naxi.rs
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: D: Nextion t2.txt="NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k"Nextion t2.txt="NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k"

D: Headerline: icy-metaint:8192
D: Headerline: icy-br:128
D: Switch to DATA, bitrate is 128, metaint is 8192
D: Duration mp3loop 40
D: Metadata block 64 bytes
D: Streamtitle found, 48 bytes
D: StreamTitle=' Mariah Carey - Hero';StreamUrl='';
D: Nextion t1.txt=" Mariah Carey\rHero"
D: NEXTION command seen 70 volume=75
D: Command: volume with parameter 75
D: Metadata block 80 bytes
D: Streamtitle found, 64 bytes
D: StreamTitle=' James Arthur - Say You Won't Let Go';StreamUrl='';
D: Nextion t1.txt=" James Arthur\rSay You Won't Let Go"

Edzelf, 2020-03-31:
Some SD cards disturb the SPI bus.  Search for the solution in other issues.

Jefke1, 2020-04-11:
Hi Ed, I mounted another SD card reader.
Playing MP3s, but artist and title are displayed for a fraction of a sec in t1 of the Nextion display. In the web interface is everything ok.
D: NEXTION command seen 70 uppreset=1
D: Command: uppreset with parameter 1
D: SD_currentnode is 0, curnod is 0, delta is 1
D: getSDfilename random choice
D: getSDfilename requested node ID is 2,2,0,0
D: Selected file is localhost/Mp3/Dido - Thank You (Acoustic).mp3
D: Playing localhost/Mp3/Dido - Thank You (Acoustic).mp3
D: STOP requested
D: Found ID3 info
D: Nextion t0.txt="ESP32 MP3 Player"
D: Nextion t1.txt=""
D: Nextion t2.txt="Playing from local file"
D: Song stopped correctly after 20 msec
D: NEXTION command seen 70 downpreset=1
D: Command: downpreset with parameter 1
D: SD_currentnode is 0, curnod is 0, delta is -1
D: getSDfilename random choice
D: getSDfilename requested node ID is 2,1,0,0
D: Selected file is localhost/Mp3/David Gilmour Wish You Were Here  1080p HD.mp3
D: Playing localhost/Mp3/David Gilmour Wish You Were Here  1080p HD.mp3
D: STOP requested
D: Found ID3 info
D: Nextion t0.txt="ESP32 MP3 Player"
D: Nextion t1.txt=""
D: Nextion t2.txt="Playing from local file"
D: Song stopped correctly after 20 msec

thanks in advance

Jefke1, 2020-04-18:
Solved ;-)

closed 2020-04-18 by Jefke1