Edzelf/ESP32-Radio #312: radio livestream with m3u ?

dl6dbh, 2020-04-11:
hallo 
i have the problem that livestream links  mp3 and m3u not working 
in the monitor d:no data input 
i havent an idea for the mistake. 
anyone can help me ?

MHz000, 2021-02-23:
same problem m3u streams not accepted like one of my favored stations

http://streams.br.de/br-klassik_2.m3u    # BR-Klassik (Klassik)
Regards

Edzelf, 2021-02-23:
Does not look like a m3u file, Try:
`br-brklassik-live.cast.addradio.de/br/brklassik/live/mp3/mid`

MHz000, 2021-02-24:
Great, thank-you Ed.
Please tell me how you figured it out.

Edzelf, 2021-02-24:
Download the file http://streams.br.de/br-klassik_2.m3u and read it.  The trick with winamp (see documentation) will probably work as well.

MHz000, 2021-02-24:
Thanks, I'll check it out.

Edzelf, 2021-04-25:
.m3u will probably work now.