Edzelf/ESP32-Radio #313: setRotation is ineffective

andreaconsole, 2020-04-12:
I'm not sure I can say this is an issue related to your software (which, by the way, is incredible: kudos!). However, I'm using an ST7735 cheap clone and I can't put the screen in landscape mode (out of the box, it works in portrait mode with your software, and the rows below the 128th are apparently covered with random noise).
I see that you used setRotation ( 1 ) for this purpose, but apparently it doesn't work with my driver/hardware. Changing the source code SetRotation parameter is ineffective.
I'm using:
Adafruit_ST7735_and_ST7789_Library  1.5.15
Adafruit_GFX_Library  1.7.5
Do you have any hint for me?
Thanks a lot.
Andrea

andreaconsole, 2020-04-14:
Update: since I was unsatisfied with the original library, I switched to the tft_eSPI library and aligned the code to use it. Now everything works as expected. It probably also provides additional features to explore...

DozoG, 2020-04-27:
I believe there are quite a few slightly different versions of those displays.
(different manufacturers, older ones, newer ones)
All use the same ST7735 library, but all require slightly different start up codes. 
(Those are hidden somewhere in the library)
Some libraries have options to select Green-tab or Black-tab or something, that will select a somewhat different startup internally in the library.
The tft-eSPI library is very versatile, but you still need to check a lot of options, didn't you find that annoying ? :-)

andreaconsole, 2020-04-28:
I was lucky: it worked at my first attempt :)

mikey60, 2020-04-29:
I found that the setRotation function only works up through version 1.2.8 of the Adafruit_ST7735_and_ST7789_Library with version 1.75 of the Adafruit_GFX_Library with my 2017 ST7335 display.   I am actually using a value of 3 for the setRotation because my display is mounted upside down.  The setRotation also works with version 1.08 of the Adafruit_ST7735 library which I installed before the library became the Adafruit_ST7735_and_ST7789_Library.