Edzelf/ESP32-Radio #318: #include <PubSubClient. what does that mean error

meinschatz1, 2020-04-17:


fellek, 2020-04-19:
please give more information about your Problem

fellek, 2020-04-20:
You will have to follow this thread
https://forum.arduino.cc/index.php?topic=319836.0

Please always post the output of the protocols.

meinschatz1, 2020-04-20:
> i get this message on arduino and can not verify to upload to esp32, also i get another one that says about_html.h: no such directory, both at the verify stage,thank you

https://forum.arduino.cc/index.php?topic=319836.0 the link  you posted helped an now i have uploaded the esp32 radio and it connect to wifi ,but the vs1053 gives me error , i am using an esp32 wroom module

Edzelf, 2020-04-21:
Please show the error messages from the serial ouput.

meinschatz1, 2020-04-22:
hello edzelf, i will attach sreenshots  of  my serial output ,i could not copy them for some reason.like i said the esp32radio conects and i see on display radio stations but no sound from mp3 module, the chip on module is vs1053b.thank you.
[image]
[image]
[image]
[image]
[image]
[image]

koskee, 2020-04-22:
Bad news mate, It says right in your output that you got a counterfeit vs1053... (if you look really close at the main chip on your board, it likely says vs1003, unless they started sanding the markings off of the chips that is. 
Theres a lot of fake ones out there, and afaik there's not really any reliable way to guarantee you get the right chip when you buy a board. What I did was asked Ed for a link that he used before and bought from there to find a legitimate version without the guesswork. 

It should still work for mp3 streams, but from what I remember it's not licenced for .aif files or whatever that strange Apple file format is.. It's still under patent and so the scam is that they sell you the cheaper chips for the (if you can call it that) more expensive price.

That being said, you should still be able to use it, it just won't work with any of the iheartradio streams or any other ones based on that weird file format. Mp3 is fine, and almost every other format, just not that one. (but it's kinda common in the ip radio scene)

meinschatz1, 2020-04-22:
thanks,koskee.I will look into it and try to  find a legitimate version if this one does not work.

fellek, 2020-04-22:
I had this vs1053 / vs100x thing too, but it seems to be gone after correct wiring and config oft the pins.
for now I ordered another display because of problems in connecting my OLED and SD card. I will report how it will do.