Edzelf/ESP32-Radio #319: Wifi found, device crashes and device goes back to default AP - endless loop

Gulpman, 2020-04-19:
Hi, I wanted to set up the project and start with nothing connected to the ESP32, to check the general functionality but am not successful. 

So when the device boots up it seems to write everything correctly into nvs, boots up again, seems to connect to my network correctly but then crashes and after that is not capable to connect to the net again.

When I do a reset on the ESP32 device, it connects to the network again, crashes and then again gets back to the default AP (ESP32Radio).

Is it neccessary to connect everything (VS1053, display) befor testing? 

Here's the log file stripped down to the interesting stuff:
`
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
D: Connected to MYGUESTNETWORK
D: IP = 192.168.178.149
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh13
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 71
D: Song stopped correctly after 0 msec
D: Duration mp3loop 400
D: Metadata block 96 bytes
D: Streamtitle found, 82 bytes
D: StreamTitle='NPO Radio 1 - De Perstribune: Paul de Leeuw en Sarina Wiegman - MAX';
D: SPI semaphore not taken within 270 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 280 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 290 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 300 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 310 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
[LEAVING OUT THE 10 TICKS INTERVALS UNTIL SOMETHING DIFFERENT IN THE LOG]
D: SPI semaphore not taken within 740 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 750 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
D: SPI semaphore not taken within 760 ticks by CPU 1, id hspec
D: Semaphore is claimed by chunk
[... 128 lines cut ...]
Writing at 0x0006c000... (77 %)
Writing at 0x00070000... (80 %)
Writing at 0x00074000... (83 %)
Writing at 0x00078000... (87 %)
Writing at 0x0007c000... (90 %)
Writing at 0x00080000... (93 %)
Writing at 0x00084000... (96 %)
Writing at 0x00088000... (100 %)
Wrote 878496 bytes (493416 compressed) at 0x00010000 in 8.2 seconds (effective 855.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1365.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

I have added my network to the defaultprefs.h for convenience and to be sure there is no type but that has the same effect.

Any idea what I am doing wrong?
Best,
Sascha

Edzelf, 2020-04-19:
Defaultprefs.h has no effect on the wifi connection.  It is only used when choosing the default setting in the web interface.
The crash can be caused by missing hardware, but this should not effect your NVS data.

fellek, 2020-04-19:
This is the same behavior I got stuck in. 
There was an issue at espressif
Containing more specific debug Logs of this crash:

https://github.com/espressif/arduino-esp32/issues/3440

Gulpman, 2020-04-20:
> 
> 
> Defaultprefs.h has no effect on the wifi connection. It is only used when choosing the default setting in the web interface.
Yes, I did not correctly express: I added it to Defaultprefs.h to have my credentials handy when pressing the default button to ensure there is no typo. Therefore for me it had the indirect effect on wifi settings to reduce possible error sources. :)

> The crash can be caused by missing hardware, but this should not effect your NVS data.
So I will try with he VS1053 connected.

Gulpman, 2020-04-20:
> 
> 
> This is the same behavior I got stuck in.
> There was an issue at espressif
> Containing more specific debug Logs of this crash:
> 
> [espressif/arduino-esp32#3440](https://github.com/espressif/arduino-esp32/issues/3440)

I wanted to downgrade the ESP32 package in the IDE but there is no single package. Any idea which package I should downgrade to test if this is the cause?
The code snipplet on the thread you are referring to seems to use Wifi and WifiMulti. I didn't find out how to downgrade those. 
Or is there a better solution to the issue?
Any clue?
Thanks,
Sascha

Gulpman, 2020-04-21:
> 
> 
> Defaultprefs.h has no effect on the wifi connection. It is only used when choosing the default setting in the web interface.
> The crash can be caused by missing hardware, but this should not effect your NVS data.

OK, fist tests can be run, if the VS1053 is connected (and no other hardware). Without the VS1053 (therefore only the ESP32 with the sketch) I could reproduce the issue on another ESP32 board.

So - issue is solved, however it would be helpful to add this to the documentation, as I think it is natural to test first with the ESP32 only and the compiled sketch running. 

Thanks for your help!

Best, Sascha

closed 2020-04-21 by Gulpman

Edzelf, 2020-04-21:
So I took the ESP32 (WROOM-32) out of the radio and powered it up.
Lots of errors, but no crash.  Web interface works.
Maybe it depends on the defined pins.
My pins:
```
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
```

Gulpman, 2020-04-22:
> So I took the ESP32 (WROOM-32) out of the radio and powered it up.
> Lots of errors, but no crash. Web interface works.
> Maybe it depends on the defined pins.
> My pins:
>
> ```
> pin_ir = [...]

Thanks for verifying. 
The only thing I changed in the original defaul config, was adding my WLAN network. 
Here's the corresponding part for the pins from my config:

`
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
`
I can see in the config you posted, that you altered the pin_vs_dcs to 16 (in the original config it's 32). Apart from that our configs seem to be identically.

I'm using the latest version of Arduino IDE (1.8.12), all libraries and board definitions updated to the latest version. 

Anyway, it runs fine now, (with the OLED connected) - only issue I'm having is that the display is not correctly displaying the content but this seems to be an issue with the chip (SSH1106). Will open an extra case for that. :)

Thanks again for investigating!

Best,
Sascha