Edzelf/ESP32-Radio #322: VS1003 support

20pyro00, 2020-04-23:
Hello,

I accidentaly bought a VS1003 board. I wired everything up, but there is no sound. I also get the error "Song stopped incorrectly!". Is the VS1003 board not working or do i have to change the code? Well if it´s not supported, i have to wait for the VS1053 board to be delivered from China :(

LOG:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 294100
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 45 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to 33
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = 212.83.149.66:8591/stream
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added Repeater2 to list of networks
D: Added NETGEAR56 to list of networks
D: Added TOBISOnePlus6 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - Repeater2                 Signal: -78 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - FRITZ!Box 7560 RA         Signal: -89 dBm, Encryption WPA2_PSK, 
D:  3 - Loitsche_A                Signal: -90 dBm, Encryption WPA2_PSK, 
D:  4 - DIRECT-8b-HP M477 LaserJet Signal: -90 dBm, Encryption WPA2_PSK, 
D:  5 - Telekom_FON               Signal: -91 dBm, Encryption OPEN, 
D:  6 - HP-Print-58-Officejet 4630 Signal: -93 dBm, Encryption OPEN, 
D:  7 - WLAN-NN5Y5H               Signal: -93 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: gpio_14 with parameter stop
D: Command: gpio_17 with parameter resume
D: Command: gpio_34 with parameter station = 212.83.149.66:8591/stream
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 149.56.151.181:9992/stream
D: Command: preset_01 with parameter 78.31.65.20:8080/dance.mp3
D: Command: preset_02 with parameter 51.89.148.171:8022/stream
D: Command: preset_03 with parameter 159.89.121.179:80/192k
D: Command: preset_04 with parameter 145.239.91.167:8010
D: Command: preset_05 with parameter 37.187.93.104:8589/deep
D: Command: preset_06 with parameter 5.39.71.159:8704/stream
D: Command: preset_07 with parameter 5.39.71.159:8243/stream
D: Command: preset_08 with parameter 212.83.149.66:8591/stream
D: Command: preset_09 with parameter 136.243.4.79:8055/stream
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: This is not a VS1053, but a VS1003 instead!
D: Connect to WiFi
D: Connected to Repeater2
D: IP = 192.168.1.10
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 12:15:55
D: STOP requested
D: New preset/file requested (6/0) from 5.39.71.159:8704/stream
D: Connect to new host 5.39.71.159:8704/stream
D: Connect to 5.39.71.159 on port 8704, extension /stream
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Accept-Ranges:none
D: Headerline: Access-Control-Allow-Origin:*
D: Headerline: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Headerline: Connection:close
D: Headerline: icy-name:DI Radio Digital Impulse - Global Trance Channel
D: Headerline: icy-genre:Electronic
D: Headerline: icy-br:320
D: Headerline: icy-sr:44100
D: Headerline: icy-url:http://www.di-radio.com
D: Headerline: icy-pub:1
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-metaint:8192
D: Headerline: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 320, metaint is 8192
D: Duration mp3loop 46
D: Metadata block 64 bytes
D: Streamtitle found, 59 bytes
D: StreamTitle='ATLAS CORPORATION - Into The Dreamland [Roj]';
D: SPI semaphore not taken within 270 ticks by CPU 1, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 300 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 310 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 320 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 330 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D: SPI semaphore not taken within 340 ticks by CPU 1, id hspec
D: Semaphore is claimed by stopsong
D:   0 -   808
D:   1 -    33
D: SPI semaphore not taken within 350 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   2 -     0
D:   3 -     0
D: SPI semaphore not taken within 360 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   4 -     0
D:   5 -  1F40
D: SPI semaphore not taken within 370 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsongD:   6 -     0

D:   7 -     0
D:   8 -     0
D: SPI semaphore not taken within 380 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   9 -     0
D:   A -     0
D: SPI semaphore not taken within 390 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   B -  F8F8
D:   C -     0
D: SPI semaphore not taken within 400 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsongD:   D -     0

D:   E -     0
D:   F -     0

Thanks
Tobias

Edzelf, 2020-04-23:
You may try to delete line 906 ( "okay = false ").  But please read other issues on using the VS1003.

20pyro00, 2020-04-23:
Oh wow! Thanks for the fast answer. And with the code changed it works! Thank you so much. But now i have a new problem. The Audio is very choppy. I think it is Network related, but i am not sure. Is there a way to increase the streanght of the signal?

Here are examples:
https://youtu.be/VBua80It10A
https://youtu.be/dXYTvYK-Dpw

When i load a 320kbps Stream, i get this:

D: Command client available
D: handlehttp started
D: Get command is: station=radiocp.from-dc.com:1290
D: Send reply for station=radiocp.from-dc.com:1290
D: Command: station with parameter radiocp.from-dc.com:1290
D: STOP requested
D: Stopping client
D: Connect to new host radiocp.from-dc.com:1290
D: Connect to radiocp.from-dc.com on port 1290, extension /
D: Connected to server
D: Switch to HEADER
D: Headerline: Accept-Ranges:none
D: Headerline: Access-Control-Allow-Origin:*
D: Headerline: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Headerline: Connection:close
D: Headerline: icy-name:deep wave - deep/dub/electronic/indie - https://deepwaveradio.fr/
D: Headerline: icy-genre:Electronic
D: Headerline: icy-br:320
D: Headerline: icy-sr:44100
D: Headerline: icy-url:https://www.deepwaveradio.fr
D: Headerline: icy-pub:1
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-metaint:32768
D: Headerline: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 320, metaint is 32768
D: SPI semaphore not taken within 270 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 300 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 310 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 320 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 330 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 340 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 350 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 360 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 370 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 380 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 390 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 400 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 410 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D: SPI semaphore not taken within 420 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   1 -    33

D:   2 -     0
D:   3 -  6000
D: SPI semaphore not taken within 430 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   4 -   116
D:   5 -  AC44
D: SPI semaphore not taken within 440 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   6 -  10C2
D:   7 -  1E06
D: SPI semaphore not taken within 450 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   8 -     0

D:   9 -     0
D:   A -     0
D: SPI semaphore not taken within 460 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   B -  F8F8
D:   C -     0
D: SPI semaphore not taken within 470 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D:   D -     0
D:   E -     0
D: SPI semaphore not taken within 480 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD:   F -     0

What are all the semaphore-tick-things?

Thanks
Tobias

closed 2020-04-24 by 20pyro00