Edzelf/ESP32-Radio #323: Very Choppy Audio

20pyro00, 2020-04-24:
Hello there. I have a new problem. The Audio is very choppy. I think it is Network related, but i am not sure.

Here are two examples:
https://youtu.be/VBua80It10A
https://youtu.be/dXYTvYK-Dpw

I noticed when i click "test" on the esp-radio-website it sometimes says "Free memory is 194448, chunks in queue 0, stream 0, bitrate 44 kbps" and there is a lag. When the stream continues, the chunks in queue go up to 400 and stream goes up to ~4000. Can i increase the chunks in queue so when there is a lag, enough chunks are loaded?

Also when i load a 320kbps Stream, i get this:

D: Command client available
D: handlehttp started
D: Get command is: station=radiocp.from-dc.com:1290
D: Send reply for station=radiocp.from-dc.com:1290
D: Command: station with parameter radiocp.from-dc.com:1290
D: STOP requested
D: Stopping client
D: Connect to new host radiocp.from-dc.com:1290
D: Connect to radiocp.from-dc.com on port 1290, extension /
D: Connected to server
D: Switch to HEADER
D: Headerline: Accept-Ranges:none
D: Headerline: Access-Control-Allow-Origin:*
D: Headerline: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Headerline: Connection:close
D: Headerline: icy-name:deep wave - deep/dub/electronic/indie - https://deepwaveradio.fr/
D: Headerline: icy-genre:Electronic
D: Headerline: icy-br:320
D: Headerline: icy-sr:44100
D: Headerline: icy-url:https://www.deepwaveradio.fr
D: Headerline: icy-pub:1
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-metaint:32768
D: Headerline: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 320, metaint is 32768
D: SPI semaphore not taken within 270 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 300 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 310 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 320 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 330 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 340 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 350 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 360 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 370 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 380 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 390 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 400 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: SPI semaphore not taken within 410 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: Song stopped incorrectly!
D: REG Contents
D: --- -----
D: 0 - 808
D: SPI semaphore not taken within 420 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD: 1 - 33

D: 2 - 0
D: 3 - 6000
D: SPI semaphore not taken within 430 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: 4 - 116
D: 5 - AC44
D: SPI semaphore not taken within 440 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: 6 - 10C2
D: 7 - 1E06
D: SPI semaphore not taken within 450 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD: 8 - 0

D: 9 - 0
D: A - 0
D: SPI semaphore not taken within 460 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: B - F8F8
D: C - 0
D: SPI semaphore not taken within 470 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsong
D: D - 0
D: E - 0
D: SPI semaphore not taken within 480 ticks by CPU 0, id hspectft
D: Semaphore is claimed by stopsongD: F - 0

What are all the semaphore-tick-things?

Thanks
Tobias

Edzelf, 2020-04-24:
There is not enough traffic from the sender to the ESP32.  Use another network or try other stations.

Edzelf, 2020-04-24:
The station radiocp.from-dc.com:1290 is working well here (although hard for me to listen to).
Test results:
`Free memory is 200356, chunks in queue 400, stream 4736, bitrate 320 kbps`

20pyro00, 2020-04-24:
> 
> 
> The station radiocp.from-dc.com:1290 is working well here (although hard for me to listen to).
> Test results:
> `Free memory is 200356, chunks in queue 400, stream 4736, bitrate 320 kbps`

Mhh ok. And is the station 212.83.149.66:8591/stream working for you? This station is very choppy.
Is there a way to increase the chunks in queue?

Edzelf, 2020-04-24:
Yes, "212.83.149.66:8591/stream" works perfect  (although very hard for me to listen to).  The results:
`Free memory is 201468, chunks in queue 394, stream 5124, bitrate 319 kbps`
You can increase the queue on line 191:
`#define QSIZ 400`
But it will not help.  Just fix your network.

20pyro00, 2020-04-24:
Ok thank you.
When i listen to the stream over my smartphone, everything is ok. So maybe the wifi-connection of the esp32 is bad?

20pyro00, 2020-04-25:
Well i closed the project. The Audio was to bad sadly :(

closed 2020-04-25 by 20pyro00