Edzelf/ESP32-Radio #324: VS1053 not working and no debug serial

meerweten, 2020-05-10:
So i just got this project from a colleague who couldn't get it working.
Guess what, neather can i.

I wired it up as in the INO file and use a nextion screen (selected the correct screen using the define)
i can acces the web interface, push the buttons on there but there is no sound comming from the vs1053 (using headphones directly to the VS1053)

So i tried connecting a serial to usb device i made myself (and is working) set up a serial program on my pc (settings 115200 bps 8N1 no handshaking) but can't get debug data on TX/RX0

Displays keeps saying connecting ...

any toughts?

on the ESP32 board only the red led is lit.

So i also connected a logic analyzer to the spi line, getting no clock... (SCK on pin 18)

Edzelf, 2020-05-11:
Try to run the ESP32 without any wiring (but USB connected for serial I/O).  Now you must have serial output.  The web interface should be working and you will be able to do the configuration.

meerweten, 2020-05-13:
thanks for the input,
Everything looks to be going ok until this point:

here i found (thanks to the debug) a dubble pin configuration
Now i got sound but its still nog showing anything on te screen

```
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 71
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Playtask) 
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 294008
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Init Nextion, pins 16,17
D: Pin 16 is already reserved!
D: Nextion command cls BLACK
D: Nextion command page 0
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added Orange-a833c to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 9
D:  1 - Guest-Orange-a833c        Signal: -65 dBm, Encryption WPA2_PSK, 
D:  2 - Orange-a833c              Signal: -66 dBm, Encryption WPA2_PSK, Acceptable
D:  3 - Orange-a833c              Signal: -81 dBm, Encryption WPA2_PSK, Acceptable
D:  4 - Guest-Orange-a833c        Signal: -81 dBm, Encryption WPA2_PSK, 
D:  5 - TelenetWiFree             Signal: -88 dBm, Encryption ????, 
D:  6 - telenet-F9EA564           Signal: -91 dBm, Encryption WPA2_PSK, 
D:  7 - Orange-9f48d              Signal: -92 dBm, Encryption WPA2_PSK, 
D:  8 - telenet-1F2CD             Signal: -92 dBm, Encryption WPA2_PSK, 
D:  9 - TelenetWiFree             Signal: -94 dBm, Encryption ????, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 SPI error. SB:004B R1:004B R2:03FF
D: VS1053 SPI error. SB:0054 R1:0000 R2:0054
D: VS1053 SPI error. SB:005A R1:0300 R2:005A
```

meerweten, 2020-05-13:
So for some reason the volume keeps going down...
Screen not updating...

Edzelf, 2020-05-14:
There is a problem with the SPI bus.  Check wiring, configuration and power supply.
The volume changes probably because the pins you have configured for the rotary encoder are open.  Ground them if you do not use them.

meerweten, 2020-05-14:
So i got the sound working,
the screen
And the volume doesn't decrese anymore :)
I'm very thankfull!

1 problems remains, my colleague (who i'm helping to get this working) bought a larges screensize from nextion then you implemented.
I got the screen working (after installing the nextion software and resizing) but it seems there is only Display information and no real code in it?
i'm asking as since i moved and rezised everything 2 of the buttons are no longer working

Edzelf, 2020-05-15:
Yes, the nextion interface is very simple.  Only textlines and buttons.  The 4 buttons on top should however cause events on release.  The commands generated by the release of the button are "stop", "mute", "uppreset=1" or "downpreset=1".  Check this in the Nextion editor.  At runtime, the commands should be visible in the debug logging.  You may test the commands by inputting them on the serial interface.