Edzelf/ESP32-Radio #326: adding radio stations

drunagr, 2020-05-10:
hi
how can i find out the full address of a station?
for example how can i add this station ? : http://www.skairadio.gr/  
or this one : http://live24.gr/radio/realfm.jsp
i have tried the winamp but no luck...

thanks

drunagr, 2020-05-11:
or from this site : https://tunein.com/radio/Real-FM-978-s78253/
i notice that all your station have the port too...

Edzelf, 2020-05-11:
Could it be "realfm.live24.gr/realfm".  Google is your friend.

drunagr, 2020-05-11:
Dear Ed 
This Link works but how did you found it?
You said google is our friend , ok but what am I looking for? I mean why the : realfm.live24.gr/realfm
Works and the “original“ one : http://live24.gr/radio/realfm.jsp
Not ?
Is the port not always necessary?
Sorry for all this questions but I have tried a lot of radio projects an no one explains where this address come from...I also searched here and it seems that no one hat this problem and I am feeling a little stupid.
Thanks for your help

Edzelf, 2020-05-11:
I do not remember what term I used to google around, but it had someting to do with "shoutcast" and "realfm".  I found `"https://github.com/ubuntu-gr/eradio-playlist-generator/blob/master/eradio.radiostations.txt"`.  That's a list with many stations.  I picked the one with "realfm" in it.
A port is necessary if it is not "80".  "80" is the default.

drunagr, 2020-05-11:
thanks a lot Ed.
i am waiting the mp3 encoder and i will build the project.

drunagr, 2020-05-20:
ok i found it!
http://fmstream.org/index.php

closed 2020-05-20 by drunagr