Edzelf/ESP32-Radio #329: Read and Play Mp3 over USB with USB Host Modul CH376

Inrouder, 2020-05-17:
Hello,

we are a group of infomatic elektirc students and we like your project!

This Esp32 Radio project is very useful for our project.
 

We have to build a HIFI system with multiple functions.
One function of this is the MP3 function with ID3 TAG and USB.
We don't want to use a Raspberry PI and co. It should only work via microcontrollers.

Now we have discovered the USB host module CH376. For this we also have a working library that is similar to the SD library, but we won't get any further because we all have little programming knowledge.

Would it be possible for you to help us with this problem?

Would also be very interested in your project.

Best regards

Peter, Janik, Steven and Werner

Edzelf, 2020-05-18:
I'm not sure if I can be of any help.  Bute send me your documentation and a description of the problem and I will see what I can do.  My mailaddress is in about_html.h.

utw3v0, 2020-05-18:
We would like to add USB - Mass-Storage functionality, so we can play Music from USB-Sticks and or USB-Hard drives within your ESP32 Radio with ID3-Tag functionality ofc. 

Our Problem is to fit in the USB to SPI Controller ( CH376S ) [CH376DS1.pdf](https://github.com/Edzelf/ESP32-Radio/files/4645034/CH376DS1.pdf) so we can use our Mass Storage device to play mp3 from it like we can do already from SD - Card. 

[image]
We appreciate your help very much, Thank you

Best regards

Peter, Janik, Steven and Werner

Edzelf, 2020-05-18:
Connecting this thing to the SPI bus look simple.  But some software is needed to read files from the USB stick.  I can write the software, but I need a CH376S to test it.  I ordered one yesterday, but it can take a month or so before it arrives here.

utw3v0, 2020-05-18:
Hello Ed, 

We could send you our ch376s module for testing and coding and you could keep it bc in our group we would directly use the ic and create an Mainboard layout, bc the usb part is only one of a lot of other features we need in our project. 

Best regards, 

Werner 

⁣TypeApp for Android herunterladen ​

Edzelf, 2020-05-18:
That's okay.  Mail me on my private mail address and I will reply with my house address.