Edzelf/ESP32-Radio #330: requested online radio failed

mmiizzoo, 2020-05-19:
hi all,

first of all, it's a nice project, I'm really amazed.
I fixed most issuses I had , I added st7789 display driver as well, mp3 from sd card works fine but the internet radio doesn't want to work. Wifi connection ok.
It tries to connect but after few sec there is a connection failed message.
can you help with this?
I tried other internet radio code and it is streaming.
so I guess something going on here.

Thanks

Edzelf, 2020-05-19:
Show the logging please.
If you remove all the wiring, you should see the connection and the stream.  Than reconnect the wiring (step by step if possible).

mmiizzoo, 2020-05-19:
When I tried the other code I didn't touch the connections I just uploaded different code so I guess it's something else but I'll try it later and I'll send the logging but now I'm at work.
thanks for quick response

mmiizzoo, 2020-05-19:
I disconnected all the wires and still same:

D: Starting ESP32-radio running on CPU 1 at 160 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 252668
D: Display type is ST7789
D: Partition nvs found, 20480 bytes
D: Read 40 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added TALKTALK6D57DA to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 4
D:  1 - TALKTALK6D57DA            Signal: -44 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - EE-BrightBox-6c69ff       Signal: -71 dBm, Encryption WPA_WPA2_PSK, 
D:  3 - VM3244167                 Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - Virgin Media              Signal: -91 dBm, Encryption ????, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 22
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 80
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
E (5883) wifi: esp_wifi_scan_start 1245 wifi not start
E (5884) wifi: esp_wifi_disconnect 1153 wifi not start
E (5884) wifi: esp_wifi_scan_start 1245 wifi not start
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is enabled
D: STOP requested
D: Song stopped correctly after 0 msec
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Command client available
D: handlehttp started
D: Get command is: preset=5
D: Send reply for preset=5
D: Command: preset with parameter 5
D: STOP requested
D: New preset/file requested (5/0) from us1.internet-radio.com:8105
D: Connect to new host us1.internet-radio.com:8105
D: Connect to us1.internet-radio.com on port 8105, extension /
D: Request us1.internet-radio.com:8105 failed!
D: Duration mp3loop 1
D: Song stopped correctly after 0 msec
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Command client available
D: handlehttp started
D: Get command is: preset=3
D: Send reply for preset=3
D: Command: preset with parameter 3
D: STOP requested
D: New preset/file requested (3/0) from airspectrum.cdnstream1.com:8000/1261_192
D: Connect to new host airspectrum.cdnstream1.com:8000/1261_192
D: Connect to airspectrum.cdnstream1.com on port 8000, extension /1261_192
D: Request airspectrum.cdnstream1.com:8000/1261_192 failed!
D: Song stopped correctly after 0 msec

mmiizzoo, 2020-05-19:
this is when all connected:
D: Starting ESP32-radio running on CPU 1 at 160 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 252668
D: Display type is ST7789
D: Partition nvs found, 20480 bytes
D: Read 40 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is LOW, probably no PULL-UP
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 35 for IR
D: Start display
D: Locate mp3 files on SD, may take a while...
D: SD directory is /
D: 3 tracks on SD
D: Create list with acceptable WiFi networks
D: Added TALKTALK6D57DA to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - TALKTALK6D57DA            Signal: -35 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - EE-BrightBox-6c69ff       Signal: -77 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 22
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 80
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
E (6755) wifi: esp_wifi_scan_start 1245 wifi not start
E (6756) wifi: esp_wifi_disconnect 1153 wifi not start
E (6756) wifi: esp_wifi_scan_start 1245 wifi not start
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is enabled
D: STOP requested
D: Song stopped correctly after 0 msec
D: Command client available
D: handlehttp started
D: Get command is: preset=4
D: Send reply for preset=4
D: Command: preset with parameter 4
D: STOP requested
D: New preset/file requested (4/0) from airspectrum.cdnstream1.com:8008/1604_128
D: Connect to new host airspectrum.cdnstream1.com:8008/1604_128
D: Connect to airspectrum.cdnstream1.com on port 8008, extension /1604_128
D: Request airspectrum.cdnstream1.com:8008/1604_128 failed!
D: Duration mp3loop 1
D: Song stopped correctly after 0 msec
D: Command client available
D: handlehttp started
D: Get command is: preset=7
D: Send reply for preset=7
D: Command: preset with parameter 7
D: STOP requested
D: New preset/file requested (7/0) from 205.164.62.15:10032
D: Connect to new host 205.164.62.15:10032
D: Connect to 205.164.62.15 on port 10032, extension /
D: Song stopped correctly after 0 msec
D: Request 205.164.62.15:10032 failed!
D: STOP requested
D: New preset/file requested (8/0) from skonto.ls.lv:8002/mp3
D: Connect to new host skonto.ls.lv:8002/mp3
D: Connect to skonto.ls.lv on port 8002, extension /mp3
D: Request skonto.ls.lv:8002/mp3 failed!
D: Song stopped correctly after 0 msec
D: STOP requested
D: New preset/file requested (9/0) from 94.23.66.155:8106
D: Connect to new host 94.23.66.155:8106
D: Connect to 94.23.66.155 on port 8106, extension /
D: Song stopped correctly after 0 msec

mmiizzoo, 2020-05-20:
I see the problem, it is the wifi connection. During init the screen doesn't say the wifi connection failed, so I thought it's ok, anyway I don't know why it fails.
Without wiring can't connect either so it's not the high current.
With other radio code, with the same wiring it connects and streams the radio, there is sound on speaker so the circuit is ok.
Something is with this code.

Edzelf, 2020-05-20:
There is no wifi_xx configuration in your set-up.  So the radio will act as an Access Point.  Connect to the access point and 192.168.4.1 and complete your configuration.

mmiizzoo, 2020-05-20:
I've done the wifi setup 100x already. I've changed it in the defaultprefs, I've changed it on the AP config page (saved, restarted), even I multiplied the same wifi name and password 
wifi_00 = mywifiname/mypassword
wifi_01 = mywifiname/mypassword
I'm sure I did type names passwords correctly, I double spelled it and no connection.
As I said different code on same esp32 with same circuit first time connects.

Sorry to be annoying but I'm really stuck here.

Edzelf, 2020-05-20:
Show the logging while you save your configuration.  And the configuration an the webinterface screen.

mmiizzoo, 2020-05-20:
D: Command client available
D: handlehttp started
D: Get command is: saveprefs
D: Send reply for saveprefs
D: writeprefs setstr clk_dst = 1                                          # Offset during daylight saving time (hours)
D: writeprefs setstr clk_offset = 1                                       # Offset with respect to UTC in hours
D: writeprefs setstr clk_server = pool.ntp.org                            # Time server to be used
D: writeprefs setstr gpio_00 = uppreset = 1
D: writeprefs setstr gpio_12 = upvolume = 2
D: writeprefs setstr gpio_13 = downvolume = 2
D: writeprefs setstr gpio_14 = stop
D: writeprefs setstr gpio_17 = resume
D: writeprefs setstr gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
D: writeprefs setstr ir_40BF = upvolume = 2
D: writeprefs setstr ir_C03F = downvolume = 2
D: writeprefs setstr mqqprefix = none
D: writeprefs setstr mqttbroker = none
D: writeprefs setstr mqttpasswd = *******
D: writeprefs setstr mqttport = 1883
D: writeprefs setstr mqttuser = none
D: writeprefs setstr pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
D: writeprefs setstr pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
D: writeprefs setstr pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
D: writeprefs setstr pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
D: writeprefs setstr pin_sd_cs = 22                                       # GPIO Pin number for SD card "CS"
D: writeprefs setstr pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
D: writeprefs setstr pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
D: writeprefs setstr pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
D: writeprefs setstr pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
D: writeprefs setstr pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
D: writeprefs setstr preset = 6
D: writeprefs setstr preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
D: writeprefs setstr preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
D: writeprefs setstr preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIAMI 256k
D: writeprefs setstr preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
D: writeprefs setstr preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
D: writeprefs setstr preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
D: writeprefs setstr preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
D: writeprefs setstr preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
D: writeprefs setstr preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
D: writeprefs setstr preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
D: writeprefs setstr preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
D: writeprefs setstr preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
D: writeprefs setstr toneha = 0
D: writeprefs setstr tonehf = 0
D: writeprefs setstr tonela = 0
D: writeprefs setstr tonelf = 0
D: writeprefs setstr volume = 84
D: writeprefs setstr wifi_00 = TALKTALK6D57DA/12345678
D: End of writing preferences
D: Read 45 keys from NVS

mmiizzoo, 2020-05-20:
clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqqprefix = none
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 22                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 6
preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIAMI 256k
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 87
#
wifi_00 = TALKTALK6D57DA/*******

mmiizzoo, 2020-05-20:
I modified the connection and it connects:

bool connectwifi()
{
  WiFi.begin(ssid, pass);
   while (WiFi.status() != WL_CONNECTED) {
  //  display.drawString(0,15,"   Connecting..");
    delay(500);
    Serial.print(".");
   }
//  char*      pfs ;                                      // Pointer to formatted string
//  char*      pfs2 ;                                     // Pointer to formatted string
//  bool       localAP = false ;                          // True if only local AP is left
//
//  WifiInfo_t winfo ;                                    // Entry from wifilist
//
//  WiFi.disconnect(true) ;                               // After restart the router could
//  WiFi.softAPdisconnect(true) ;                         // still keep the old connection
//  if ( wifilist.size()  )                               // Any AP defined?
//  {
//    if ( wifilist.size() == 1 )                         // Just one AP defined in preferences?
//    {
//      winfo = wifilist[0] ;                             // Get this entry
//      WiFi.begin ( winfo.ssid, winfo.passphrase ) ;     // Connect to single SSID found in wifi_xx
//      dbgprint ( "Try WiFi %s", winfo.ssid ) ;          // Message to show during WiFi connect
//    }
//    else                                                // More AP to try
//    {
//      wifiMulti.run() ;                                 // Connect to best network
//    }
//    if (  WiFi.waitForConnectResult() != WL_CONNECTED ) // Try to connect
//    {
//      localAP = true ;                                  // Error, setup own AP
//    }
//  }
//  else
//  {
//    localAP = true ;                                    // Not even a single AP defined
//  }
//  if ( localAP )                                        // Must setup local AP?
//  {
//    dbgprint ( "WiFi Failed!  Trying to setup AP with name %s and password %s.", NAME, NAME ) ;
//    WiFi.softAP ( NAME, NAME ) ;                        // This ESP will be an AP
//    pfs = dbgprint ( "IP = 192.168.4.1" ) ;             // Address for AP
//  }
//  else
//  {
//    ipaddress = WiFi.localIP().toString() ;             // Form IP address
//    pfs2 = dbgprint ( "Connected to %s", WiFi.SSID().c_str() ) ;
//    tftlog ( pfs2 ) ;
//    pfs = dbgprint ( "IP = %s", ipaddress.c_str() ) ;   // String to dispay on TFT
//  }
//  tftlog ( pfs ) ;                                      // Show IP
//  delay ( 3000 ) ;                                      // Allow user to read this
//  tftlog ( "\f" ) ;                                     // Select new page if NEXTION 
//  return ( localAP == false ) ;                         // Return result of connection
}

mmiizzoo, 2020-05-20:
I got it.
In the "connectwifi()" bool,  if I remove the **WiFi.disconnect(true) ;** line, it works.