Edzelf/ESP32-Radio #333: Nothing shoing on Display

etscho1st, 2020-05-19:
Hello ED., 
no i finaly could start the radio without reebots but now i get a new problem. I´am using a Chinese nextion like TJC. I created a HMI with the exactly configuration as yours but now when the esp32 is booting and starting nothing is happening. It only stay at page0 and not showing something.
Do u may know what can i try else thx? 

` D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 294196
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 38 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 32
D: pin_vs_dcs set to 33
D: pin_vs_dreq set to 35
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is LOW, probably no PULL-UP
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Start display
D: Init Nextion, pins 16,17
D: Nextion command cls BLACK
D: Nextion command page 0
D: Create list with acceptable WiFi networks
D: Added LizardSquad to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - LizardSquad               Signal: -50 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - FRITZ!Box Gastzugang      Signal: -50 dBm, Encryption WPA2_PSK, 
D:  3 - WLAN-3CUUSS               Signal: -81 dBm, Encryption WPA2_PSK, 
D:  4 - LizardSquad               Signal: -87 dBm, Encryption WPA2_PSK, Acceptable
D:  5 - FRITZ!Box Gastzugang      Signal: -88 dBm, Encryption WPA2_PSK, 
D:  6 - Telekom_FON               Signal: -90 dBm, Encryption OPEN, 
D:  7 - WLAN-302402               Signal: -92 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter 192.168.178.45
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 32
D: Command: pin_vs_dcs with parameter 33
D: Command: pin_vs_dreq with parameter 35
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to LizardSquad
D: IP = 192.168.178.40
D: Nextion command page 1
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MQTT uses prefix P2FC4F5FC
D: Init MQTT
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 22:08:16
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Song stopped correctly after 0 msec
D: (Re)connecting number 1 to MQTT 192.168.178.45
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh13
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 73
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t2.txt="NPO Radio1"
D: Publish to topic P2FC4F5FC/ip : 192.168.178.40
D: Publish to topic P2FC4F5FC/icy/name : NPO Radio1
D: Publish to topic P2FC4F5FC/preset : 6
D: Duration mp3loop 312
D: Metadata block 64 bytes
D: Streamtitle found, 61 bytes
D: StreamTitle='NPO Radio 1 - Langs de Lijn En Omstreken - NOS';
D: Nextion t1.txt="NPO Radio 1\rLangs de Lijn En Omstreken - NOS"
D: Publish to topic P2FC4F5FC/icy/streamtitle : NPO Radio 1 - Langs de Lijn En Omstreken - NOS
D: Publish to topic P2FC4F5FC/playing : 1
D: Publish to topic P2FC4F5FC/volume : 72
`

etscho1st, 2020-05-23:
No one any idea what can i try? Radio is working fine.

Edzelf, 2020-05-23:
Your logging is hard to read.  I had to insert newlines to read it.
D: Init Nextion, pins 16,17
D: Nextion command cls BLACK
D: Nextion command page 0
.
D: Nextion command page 1
.
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t2.txt="NPO Radio1"
.
D: Nextion t1.txt="NPO Radio 1\rLangs de Lijn En Omstreken - NOS"

So the commands to Serial 2 are generated.

Have you connect it to Serial 2 (RX2 TX2 pins)?

etscho1st, 2020-05-23:
Sorry i will try to change log. Yes i use the 16 and 17 pin like in the pictures. 
[image]

closed 2020-05-23 by etscho1st