Edzelf/ESP32-Radio #335: vs1001 instead vs1053

drunagr, 2020-05-20:
hi ED
everything seems to work fine except no SOUND . Any ideas? Thanks a lot
[image]

drunagr, 2020-05-20:
i tried to comment the line 906 //okay = false ;
but no sound again...

drunagr, 2020-05-20:
oh , on the vs1053 board i have one 5v connection not in use, is it correct?

drunagr, 2020-05-20:
i have a rotary and a oled screen on...

etscho1st, 2020-05-20:
i have to connect both 5v pins on my vs1053. for me its not running with only one

drunagr, 2020-05-20:
@etscho1st thanks! Does it need any resistors? Or just at 5 v both of them?

drunagr, 2020-05-20:
no , no sound!

drunagr, 2020-05-22:
[image]
[image]

drunagr, 2020-05-22:
GPIO19 is LOW, probably no PULL-UP
how can i fix that?

etscho1st, 2020-05-22:
This are my vs1053 settings. Only this is working at me. 
pin_vs_cs =  32                                     # GPIO Pin number for VS1053 CS
pin_vs_dcs = 33                                     # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 35                                    # GPIO Pin number for VS1053 DREQ
#

Edzelf, 2020-05-22:
Pin 19 is the SPI MISO line.  It it probably grounded by bad wiring or bad SD card interface.

drunagr, 2020-05-22:
hi Ed
now pin 19 seems ok but no sound...
any ideas, please...?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
......
.......
[image]

Edzelf, 2020-05-22:
You have selected a station with AAC encoding.  This will not work on a VS1003.

drunagr, 2020-05-22:
hi Ed 
but the last one is: audio/mpeg . This one should go ,no?

blotfi, 2020-05-24:
from your config (local server)
delete all these lines :

```
# Example configuration
# Programmable input pins:
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
```
perhaps you have one input that is catching Electromagnetic Interference and mute or stop it !
also try a radio that you are sure of its MP3 format like these 3 ones
dont put the http://

```
preset_00 = energycharts.ice.infomaniak.ch/energycharts-high.mp3   #  0 - energycharts
preset_01 = uk2.internet-radio.com:8024/ #  1 - DanceUK
preset_02 = direct.franceinfo.fr/live/franceinfo-midfi.mp3  #  2 - franceinfo

```