Edzelf/ESP32-Radio #336: isr_IR() is fired but no ir_value

blotfi, 2020-05-24:
Program is working great, 
Now I would like to add IR support
I connected an IR receiver (1838 for use with 38 KHz IR signals) for Arduino
on GPIO35
isr_IR() is fired but no ir_value, so no code available on the debugger monitor
Are there special IR remote (I tried 2)
any clue ?
thanks

Edzelf, 2020-05-24:
You can not put dbgprint lines in interrupt routines.
I have tested several remotes with success, but maybe your remote does not emit 65 pulses.  You may try lower this number in line 1926 or try another remote.

blotfi, 2020-05-24:
> You can not put dbgprint lines in interrupt routines.
> I have tested several remotes with success, but maybe your remote does not emit 65 pulses. You may try lower this number in line 1926 or try another remote.

yes I should not but Serial print in ISR

blotfi, 2020-05-24:
I tried a very cheap remote (for a datashow bundle) at it decodes IR, even with  ir_loccount == 65 
So it is a matter of IR remote
Unfortunately, the IR remote of my old HiFi (that I would like to use as an amplifier + Speaker via its AUX input) is not supported.

Also when I connect my earphone jack it works
but with the AUX it doesn't, so I guess I need to put an amplifier like http://educ8s.tv/part/PAM8403
?

blotfi, 2020-05-24:
```
# Some IR codes
ir_18E7 = upvolume = 2
ir_847B = downvolume = 2
ir_609F = downpreset = 1
ir_10EF = uppreset = 1
ir_C03F = preset = 0
ir_20DF = preset = 2
ir_A05F = preset = 3
ir_44BB = mute

```
Here are the codes to scan presets and select directly the preset channels
You should add their syntax to the pdf document, as it is not obvious and can help users.
Thanks