Edzelf/ESP32-Radio #342: Control the Web Radio with a remote control on Android

blotfi, 2020-06-06:
Control the Web Radio with a remote control on Android:
https://play.google.com/store/apps/details?id=com.embesystems.esp_radio_rc

**No Ads, totally free, no authorisation** (only internet access to send/receive data from ESP32 web radio)

I made it for myself, for my web radios, and I share the App with all the happy users of these ESP32 / 8266 web radios.

Try my mods:
https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer
and
https://github.com/blotfi/ESP32-Radio-with-SalatTime
Enjoy

avulyanichev, 2020-09-04:
Спасибо за программу. Можете сделать вывод на экран радио название станции как в строке ---Presets.------. или подскажите как это сделать.
[image]
[image]
Спасибо.

blotfi, 2020-09-04:
> Спасибо за программу. Можете сделать вывод на экран радио название станции как в строке ---Presets.------. или подскажите как это сделать.
> [image]
> [image]
> Спасибо.

Please write in English
is it 
"Thanks for the program. You can display the radio station name as in the line --- Presets .------. or tell me how to do it."
it is already displayed in the drop list

avulyanichev, 2020-09-07:
//I found the readhostfrompref (); function in the sketch that outputs a string with the current information by the preset number (currentpreset).
//selecting the line after #
String n_presset ; // I added a variable to display the station name.
String n_presset = readhostfrompref(cur rentpreset); // 
uint8_t n_simvola = n_presset.indexOf("# ");// finding a location "#"
dsp_println (n_presset.substring (n_simvola + 2)); // +2 this means that the '#' character with a space does not appear on the screen

//tell me how to insert this data in LCD1602.h in  -----   dline[1].str = (tftdata[2].str);                     // Local copy