Edzelf/ESP32-Radio #349: VS1003

Dzarda7, 2020-06-28:
Is it still possible to use vs1003? I accidentaly bought it, tried uncomment okay=False but still does not work. Thank you so much for this project

Edzelf, 2020-06-28:
I cannot test it, for I do not have one.  Some codecs will not work, but maybe mp3 will.

Dzarda7, 2020-06-29:
And I have pin 18 set to low, what might be the problem? Thank you.

Dzarda7, 2020-06-29:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 287932
D: Display type is LCD1602
D: Partition nvs found, 20480 bytes
D: Read 47 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 22
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start display
D: Init LCD1602, I2C pins 21,22
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added WiFi_7 to list of networks
D: Added sk1 to list of networks
D: End adding networks
[... 84 lines cut ...]
D: Connect to icecast5.play.cz on port 80, extension /impuls128.mp3
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Server: Icecast 2.4.4
D: Headerline: Connection: Close
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Pragma: no-cache
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
D: Headerline: icy-br:128
D: Headerline: icy-name:Impuls
D: Headerline: icy-pub:0
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Duration mp3loop 44
D: Song stopped correctly after 0 msec
D: Duration mp3loop 201
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';

Edzelf, 2020-06-30:
There is somesting wrong with the wiring of pin 18 (SPI clk).  Remove the clockline wiring step by step and check if "D: GPIO18 is LOW, probably no PULL-UP" disapears.

Dzarda7, 2020-06-30:
I disconected it but it did not disapear

sp9mrn, 2020-08-31:
same problem - wrong buy, Chinese sell 1003 board as "1003/vs1053" - be careful. 
I will try to check today.

sp9mrn, 2020-08-31:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 21 Dec 2019 09:30:00 GMT.  Free memory 288164
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 46 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added 17230 to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 9
D:  1 - 17230                     Signal: -62 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - Najlepsze_WiFi_EVER       Signal: -67 dBm, Encryption WPA2_PSK, 
D:  3 - heksa                     Signal: -77 dBm, Encryption  WEP, 
D:  4 - HP-Print-17-LaserJet M12  Signal: -77 dBm, Encryption OPEN, 
D:  5 - SFamily                   Signal: -81 dBm, Encryption WPA2_PSK, 
D:  6 - UPC0803467                Signal: -85 dBm, Encryption WPA_WPA2_PSK, 
D:  7 - andbar                    Signal: -86 dBm, Encryption WPA2_PSK, 
D:  8 - DIRECT-6E-HP DeskJet 3630 series Signal: -87 dBm, Encryption WPA2_PSK, 
D:  9 - UPC241523794              Signal: -92 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: gpio_14 with parameter stop
D: Command: gpio_17 with parameter resume
D: Command: gpio_34 with parameter station = icecast.omroep.nl:80/radio1-bb-mp3
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: lstmods with parameter Tue, 14 Jul 2020 09:44:13 GMT
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter stream.rcs.revma.com/ypqt40u0x1zuv
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: This is not a VS1053, but a VS1003 instead!
D: Connect to WiFi
D: Connected to 17230
D: IP = 192.168.1.45
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 23:19:50
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icyot takeD: SPI semaphore not taken within 120 ticks by CPU 0, id hspec

D: Semaprline: icy-genre:Talk

donnersm, 2020-12-15:
VS1003 will work but you need to comment out a line:
  write_register ( SCI_CLOCKF, 6 << 12 ) ;              // Normal clock settings

zkarolyi, 2021-04-04:
@donnersm ! Can you share more details, how to get it work? Commenting out the line above is not working.
Thank you!

glaslos, 2021-04-05:
I have a 1003B working with just skipping the version check. The chip description is:
```
VLSI Solutions
VS1003B
8332430
1924
```