Edzelf/ESP32-Radio #350: Bigger fonts possible at “ILI9341" 240x320?

Gelegenheitsbastler, 2020-07-02:
I use a “ILI9341" 240x320 like this:
https://www.marotronics.de/28-TFT-LCD-Display-Touchscreen-SPI-240x320-ILI9341-fuer-Arduino

The font is very small on this big display. Is it possible to us a bigger font? It would be so much easier to read for old mans eyes ;-)

Edzelf, 2020-07-02:
It would not be too hard to find the font size setting in the code.

Gelegenheitsbastler, 2020-07-02:
I changed line 3617
dsp_setTextSize ( 1 )
from 1 to 2.
Everything seems to be fine, but the clock is unreadable. Also the beginning of the next line is unreadable. I took a photo:
https://www.dropbox.com/s/h44utims0pl7dat/Font%20size.jpg?dl=0

Edzelf, 2020-07-02:
Search for "TIMEPOS".  That's the position of the time relative to the right side of the display.
Then search for "void displaytime ( const char* str, uint16_t color )".  At the end you'll see:
"pos += 6 ; ".  6 is the distance between the time digits.  Change it to 10 or something like that.

Gelegenheitsbastler, 2020-07-02:
It works :-)
I changed the following lines:

#define TIMEPOS -90
dsp_fillRect ( pos, 0, 11, 22, BLACK ) ;     // Clear the space for new character
pos += 11 ;

The only thing that does not work correctly is the green line under the clock. Have a look to the photo:
https://www.dropbox.com/s/c4qk8lti339uij2/IMG_20200702_163931_copy_800x600.jpg?dl=0

hevet, 2020-10-15:
Set           dsp_fillRect ( pos, 0, 10, 16, BLACK ) ;     // Clear the space for new character

arminth, 2020-10-22:
Maybe it helps to know that the changes must be made in **ESP32_radio.ino** 
dsp_setTextSize ( 1 ) 
#define TIMEPOS -90
 and in **ILI9341.h** 
dsp_fillRect ( pos, 0, 10, 16, BLACK ) ; // Clear the space for new character
pos += 11 ;