Edzelf/ESP32-Radio #354: font

avulyanichev, 2020-07-04:
Hello Ed. Please show me where the 1602 I2C output fonts are located. I want to fix the font for displaying Russian characters (screen 1602 supports Cyrillic). I will take the font from the forum from the KA32Radio project. I really want to complete a new project for myself, I do not want to switch to another radio, I am completely satisfied with everything in Your project, the radio works without hesitation in the new version now any distance works well. Thanks.

Edzelf, 2020-07-06:
The 1602 driver uses the 1602 internal character set.  You can have user defined characters, but only for 8 different characters.  So you need a 1602 with Cyrillic font ROM.

avulyanichev, 2020-07-09:
Thank you for your answer, ED. I have a 1602 display with Cyrillic. Tell me where to fix your code. Thanks.

avulyanichev, 2020-07-09:
I can't understand the logic of displaying data on the screen. Is it in the 1602.h file?

Edzelf, 2020-07-09:
Yes, a call to tft->print() will put a character on the display.