Edzelf/ESP32-Radio #357: Radio starts normaly, than lowers the volume until silence

trombose009, 2020-07-16:
Hi, 

finally my radio works. The soldering on the PCB 1.1 is now very clean and solid. 
Now I have this issue:

As soon as the radio is switched on, it starts normaly. The display works well (160*128) and I can see the starting procedure. 
After it's connected to my wifi, it starts playing the choosen radio station. 
But after a few seconds, the volume starts fading down until you cannot hear anything anymore. I have to restart the radio to start it again and every time the radio lowers the noice after a few seconds.
(the display is still showing the time. )
what could be the issue?

Edzelf, 2020-07-16:
The radio is receiving signals to turn the volume down.  You should see this on the display.  Please show the full Serial output logging, the reason should be visible there.

trombose009, 2020-07-16:
oh, "Rotation count -1"

where does it come from?

Edzelf, 2020-07-16:
Find out what pins are assigned to the rotary encoder.  Should be in the logging.  Ground those pins.

trombose009, 2020-07-16:
`D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to fritzboxdampf
D: IP = 192.168.178.59
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 12:36:53
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Song stopped correctly after 0 msec
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh14
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 73
D: Duration mp3loop 82
D: Metadata block 64 bytes
D: Streamtitle found, 51 bytes
D: StreamTitle='NPO Radio 1 - De Nieuws BV - BNNVARA';
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
`

maybe I can disable (how?) the rotary encoder? I don't have any,

trombose009, 2020-07-16:
I solved the issue now by changing the PIN number for rotary encoder CLK to an unused pin:
pin_enc_clk = 26    

I have done this by using the webinteface

Edzelf, 2020-07-16:
Right!  The normal value for an unused pin is "-1".  In the serial logging (the part that you did not want to share) you see lines like this: