Edzelf/ESP32-Radio #358: ESP32 Radio Board

uli99, 2020-07-17:
Hi,
just let me say I really like this project - thanks for making this available!

I'm in the process of building a portable Internet-Radio Boombox and for this I designed a PCB around a ESP32-WROVER module. In case somebody is interested in the KiCAD design files or some PCBs, let me know.

[image]
[image]
[image]

Uli

truckershitch, 2021-01-15:
Hi @uli99 

I'm interested in a PCB and also the KiCAD files.  I can be reached at ppvpbhhxq@relay.firefox.com

Thanks!

uli99, 2021-01-15:
Hi,
sure, pcb including smd-audio transformers or without? Address?

I tried to sent the KiCAD-Files to ppvpbhhxq@relay.firefox.com but it failed ("This message could not be delivered due to a recipient error")...

You can download the zip-file [here](https://www.senita.de/nextcloud/index.php/s/2CreLND6KGe6DCX)

Uli

truckershitch, 2021-01-15:
Hi Uli,

I can only get very small attachments there (150 KB I think -- it's a Firefox Relay address for privacy).  It seems that GitHub users can't send private messages to each other.

I downloaded the zip file.  Thank you.

If you send me just a text email, I should receive it, and I'll send you my mailing address, etc.

uli99, 2021-01-17:
sent you a plain text mail

truckershitch, 2021-01-17:
Hmm, that's weird.  It's been a few hours and I haven't received your email.

I sent myself a test message.  It forwards to my Gmail account, and my test went through.

Do you use Discord? https://discord.com

uli99, 2021-02-09:
Hi,

sorry for the late response.

I have used it for a local Fablab here in Potsdam where I live.

Uli

truckershitch wrote on 17.01.2021 20:01 (GMT +01:00):

uli99, 2021-02-09:
Hi, 

ok I just looked and I never used Discord, I used Discord. Anyway, I created a account and a discord server:

https://discord.gg/YYnKaKKs

The link is valid for 7 days.

Uli

truckershitch wrote on 17.01.2021 20:01 (GMT +01:00):

truckershitch, 2021-02-09:
Hi Uli,

I joined your discord server -- I followed the link that you sent -- but I can't seem to speak to you yet.  There seems to be a voice channel, but I don't have a microphone or a camera set up on this machine at the moment.

I see your user id on there.  I tried, but I was unable to send a message.  Perhaps the channel needs to be set up?  I have not done this before.

So here is my server invite: https://discord.gg/KkxaWqZd

uli99, 2021-02-09:
Hi, 

seems I had to give you the permission to write messages. Hope it is working now.

Uli