Edzelf/ESP32-Radio #359: Continuous resets

patryk84a, 2020-07-19:
I have a problem with starting the radio. I only have ESP32 and VS1053 module. I don't have buttons, display and an encoder. Pins for buttons and encoder turned off in the configuration. I would like to control the radio only via the website. What's the problem? Thanks for any help.

``D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 0
9:40:00 GMT.  Free memory 289316
                                              
D: Display type is DUMMYTFT
                                                   
D: Partition nvs found, 20480 bytes
                                           
D: Read 32 keys from NVS
                                                      
D: pin_ir set to -1
                                                           
D: pin_enc_clk set to -1
                                                      
D: pin_enc_dt set to -1
                                                       
D: pin_enc_sw set to -1
                                                       
D: pin_tft_cs set to -1
                                                       
D: pin_tft_dc set to -1
                                                       
D: pin_tft_scl set to -1
                                                      
D: pin_tft_sda set to -1
                                                      
D: pin_tft_bl set to -1
                                                       
D: pin_tft_blx set to -1
                                                      
D: pin_sd_cs set to -1
                                                        
D: pin_ch376_cs set to -1
                                                     
D: pin_ch376_int set to -1
                                                    
D: pin_vs_cs set to 5
                                                         
D: pin_vs_dcs set to 32
                                                       
D: pin_vs_dreq set to 4
                                                       
D: pin_shutdown set to -1
                                                     
D: pin_shutdownx set to -1
                                                    
D: pin_spi_sck set to 18
                                                      
D: pin_spi_miso set to 19
                                                     
D: pin_spi_mosi set to 23
                                                     
D: GPIO0 is HIGH
                                                              
D: GPIO2 is HIGH
                                                              
D: GPIO4 is LOW, probably no PULL-UP
                                          
D: GPIO5 is HIGH
                                                              
D: GPIO12 is HIGH
                                                             
D: GPIO13 is HIGH
                                                             
D: GPIO14 is HIGH
                                                             
D: GPIO15 is HIGH
                                                             
D: GPIO16 is HIGH
[... 707 lines cut ...]
                                           
D: Command: volume with parameter 6
                                           
D: Slow SPI, Testing VS1053 read/write registers...
                           
D: Fast SPI, Testing VS1053 read/write registers again...
                     
D: endFillByte is 0
                                                           
D: Connect to WiFi
                                                            
D: Try WiFi Ziggo3148465
                                                      
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio
.
                                                                             
D: IP = 192.168.4.1
                                                           
D: Start server for commands
                                                  
D: Rotary encoder is disabled (-1/-1/-1)
                                      
D: STOP requested
                                                             
D: Song stopped correctly after 0 msec``

Edzelf, 2020-07-19:
As you can see: "D: GPIO4 is LOW...",  so there is something wrong with the wiring or you VS1053 board.  Pin DREQ should be HIGH normally.

patryk84a, 2020-07-19:
What could be with the VS1053 module? Connections are ok. Once in the log I have HIGH and once LOW. Pin18 is also LOW. How can I check the VS1053 module differently?

patryk84a, 2020-07-19:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 09:40:00 GMT.  Free memory 294308
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 32 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: Added Ziggo3148465 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - LENOVO-PC6667             Signal: -28 dBm, Encryption WPA2_PSK, 
D:  2 - kreuzelweg17              Signal: -73 dBm, Encryption WPA2_PSK, 
D:  3 - kw17-19                   Signal: -74 dBm, Encryption WPA2_PSK, 
D:  4 - Ziggo                     Signal: -80 dBm, Encryption ????, 
[... 576 lines cut ...]
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 6
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Try WiFi Ziggo3148465
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
D: STOP requested
D: Song stopped correctly after 0 msec
`

patryk84a, 2020-07-19:
My vs1053 module is red with a built-in SD socket and a microphone.

xP373Rx, 2021-01-09:
If you are using the  ESP32 UNO version with the red shield. The voltage should be above 7v on the DC input jack. with up to 6.5v i got resets too