Edzelf/ESP32-Radio #361: Название станций из config файла

avulyanichev, 2020-07-22:
Эд, пожалуйста, помогите мне отобразить название станции из файла конфигурации. Это значительно улучшит изображение названия станции, так как вы можете отредактировать название станции самостоятельно в файле конфигурации. Спасибо.

Aleks-Ale, 2020-09-02:
> название станции на экране из файла

Один хороший человек в содружестве с другим хорошим человеком уже решили эту задачу :))))
[here](http://www.cqham.ru/forum/showthread.php?40897-%CF%F0%EE%E5%EA%F2-%E2%FB%F5%EE%E4%ED%EE%E3%EE-%E4%ED%FF-%C8%ED%F2%E5%F0%ED%E5%F2-%F0%E0%E4%E8%EE-%ED%E0-ESP32-VS1053-ILI9341)

avulyanichev, 2020-09-04:
Блин, я лох, видел же это и не вкурил в чем суть. Спасибо ТОВАРИЩ. Да ребята хорошие по ходу. Я практически изобретал велик, а все уже есть.