Edzelf/ESP32-Radio #362: Sporadic preset change

dreamy1, 2020-07-28:
Hello Edzelf,

thank you so much for this fantastic project!

I have a (minor) bug found: after some time (occurs after a few hours playing, but not reproducible time), the radio changes itself the preset. For example, i hear preset 0 and this one is saved also in the spiffs on the config site (with save & restart), but the radio changes sporadically itself to preset 1.

The (previously saved) preset 0 is normally shown on the config site after restarting, so this is correct. The curios is that after the sporadic self-changing to preset 1 is also saved into the spiffs, so this is the new preset after restarting and visible in the config site.

Do you have any ideas about this issue?

Thanks for your help and best regards,
Stefan

Edzelf, 2020-07-28:
After some time, the preset-channel that you are listening to is automatically saved as the new default preset.

dreamy1, 2020-07-28:
Sorry for explaining that misleading: i am hearing preset 0 and this one is also saved in spiffs. But the radio changes itself to preset 1 and saves this...

So this is part of my config site:
preset = 0
preset_00 = stream.antenne.de/antenne/stream/mp3     #  0 - Antenne Bayern
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k

At the moment, i am hearing preset 0...and after a while the Florida hits are playing without doing anything with the radio.

Rainer-G, 2020-07-28:
Hello Stefan,

I had the same problem and I believe it is not a bug but a feature:

If the radio stream stops for a certain time, the software switches to the next preset. I found the same on a commercial web radio, it stops and stays silent until the stream is available again.
On Ed’s radio, it skips to the next avaiable station. The same applies if the preset selected ( up / down ) is not active. The active station is saved in memory after a certain time.

Ed, are my findings right ?

Best regards

Rainer

Von: Stefan
Gesendet: Dienstag, 28. Juli 2020 17:09
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: [Edzelf/ESP32-Radio] Sporadic preset change (#362)

Hello Edzelf,
thank you so much for this fantastic project!
I have a (minor) bug found: after some time (occurs after a few hours playing, but not reproducible time), the radio changes itself the preset. For example, i hear preset 0 and this one is saved also in the spiffs on the config site (with save & restart), but the radio changes sporadically itself to preset 1.
The (previously saved) preset 0 is normally shown on the config site after restarting, so this is correct. The curios is that after the sporadic self-changing to preset 1 is also saved into the spiffs, so this is the new preset after restarting and visible in the config site.
Do you have any ideas about this issue?
Thanks for your help and best regards,
Stefan

dreamy1, 2020-07-28:
Hi Rainer,

thanks for your information, that would be an explanation. But this "feature" has the side effect of switching the preset on its own even when there is a (maybe small) interrupt of the stream. If this is implemented that way, it would be great to have a possibility to deactivate it.

Rainer-G, 2020-07-28:
Hello Stefan,

the logic to switch to next preset seems to be in the timer10sec() interrupt routine:  if there is no data, the variable morethanonce is incremented.
If morethanonceis >0, next preset is selected, if >10, ESP32 is restarted.

So, it should not happen after a very short time as first the buffer is emptied. So, if the station comes back before the buffer is empty, there should be no aoutage.

Does it happen on your side after a very short outage? I found it on NDR1 Welle Nord and on Radio FFH, other stations seem to be more reliable!

Cheers

Rainer

Von: Stefan
Gesendet: Dienstag, 28. Juli 2020 17:44
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] Sporadic preset change (#362)

Hi Rainer,
thanks for your information, that would be an explanation. But this "feature" has the side effect of switching the preset on its own even when there is a (maybe small) interrupt of the stream. If this is implemented that way, it would be great to have a possibility to deactivate it.

dreamy1, 2020-07-28:
Hi Rainer,

wow, thanks for your great help!

I can not confirm how long the outage will be there because the radio is not always observed. But it seems you are right, with the empty buffer the next station will then be automatically switched to the next preset and then a second mechanism to save this as the new "main preset" will follow.

An empty buffer alone should not activate that in my opinion. Here it would be interesting how much time the buffer will bridge this ...so a simple IP change at 12pm of the provider with some seconds for assigning the new IP would activate this also. So it would be not the ideal solution for a 24/7 web radio (e.g. in an home automation system) if the originally stored "main preset" is changed by the radio itself. 

Maybe Edzelf has an idea to activate/deactivate this behavior of the radio (a new parameter in the config side would be great)?

Thanks again and best regards,
Stefan

utw3v0, 2020-07-29:
It's gonna safe every 10 Minutes if there are any changes see here:

> //**************************************************************************************************
//                                      H A N D L E S A V E R E Q                                  *
//**************************************************************************************************
// Handle save volume/preset/tone.  This will save current settings every 10 minutes to            *
// the preferences.  On the next restart these values will be loaded.                              *
// Note that saving prefences will only take place if contents has changed.                        *
//**************************************************************************************************

Greetings Werner

dreamy1, 2020-07-29:
Hi Werner,

thanks!

I will place a pull request....

Regards,
Stefan

utw3v0, 2020-07-29:
> Hi Werner,
> 
> thanks!
> 
> I will place a pull request....EDIT: this seems is only possible with a solution but i am not really fit enough to modify the software on my own :-(
> 
> Regards,
> Stefan

Well just go there its very good commented just put a "//" without the qoutes ofcourse in front of the lines you dont wont to be saved like this //nvssetstr , then this line will be ignored by the compiler ...

dreamy1, 2020-07-29:
Thanks Werner,

i will try this as a workaround - but with the next software update this "quick & dirty solution" will bring back the old behavior. So i have the hope that Edzelf will have a look on it whether a new parameter would be a better, long term solution for it.

Rainer-G, 2020-07-30:
Hi all,
I wonder whether it would be a good idea to remove the automatic saving of the current preset. 
Imaging you change from preset_01 to preset_09 by scrolling through the list. If you listen long enough to the new station, a restart of the radio will play preset_09. 
So do all the commercial web radios I saw so far. If the radio does not save, it will always start with preset_01 and you need to select _09 again, if this is your new favorite station.
If the radio starts to change to next station after the stream stops for more than 10 seconds and you do not want the radio to do so, either 
remove this part
   If morethanonce is >0
or change to a higher value than zero to increment in 10 seconds steps.

You can test this behavior by putting an invalid ULR into the station list using the Web interface, e.g.
  preset_01 valid URL
  preset_02 invalid URL
  preset_03 valid URL
  preset_04 valid URL
Play station on preset_01, then select next preset ( Web, rotary encoder or buttons ). The radio will go to preset_02 and, as there will be no data coming in, go to preset_03.
I have a commercial Web radio as well and this will stay silent until the stream or the WiFi connection comes back. So, this is the behavior if you remove the   **If morethanonce is >0** lines

Regards
Rainer

dreamy1, 2020-07-30:
Hi Rainer,

there was absolutely no intence to **remove** that behaviour, but creating a **new parameter** where the user can **decide between the two possibilities** (auto-change preset when the stream gets lost oder silence until the stream comes back). So both would be possible.