Edzelf/ESP32-Radio #363: no first start. reboot only

Aleks-Ale, 2020-08-17:
Dear Sirs, could you help me please. I don't understand what happened. I can't get start.
I also tried to use the AC power source.  I am afraid that I have not taken important steps to launch. Arduino IDE 1.8.13(Windows Store 1.8.42.0)
----------------
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 09:40:00 GMT.  Free memory 293136
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 18
D: pin_tft_sda set to 23
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Locate mp3 files on SD, may take a while...
D: SD directory is /
D: SD directory is /System Volume Information
D: SD directory is /222
D: SD directory is /222/data
D: SD directory is /data
D: 0 tracks on SD
D: Create list with acceptable WiFi networks
D: Added brg to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - brg                       Signal: -71 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - Lovit_325                 Signal: -84 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 22
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to brg
D: IP = 192.168.1.115
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 04:29:24
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e1e47  PS      : 0x00060230  A0      : 0x800d2788  A1      : 0x3ffb1f20  
A2      : 0x00000000  A3      : 0x3f402896  A4      : 0x3ffb1f28  A5      : 0x3ffc8fa8  
A6      : 0x3ffc8fbc  A7      : 0x00000000  A8      : 0x800e6214  A9      : 0x3ffb1e50  
A10     : 0x3ffb1eac  A11     : 0x3f402896  A12     : 0x00000001  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e1e47:0x3ffb1f20 0x400d2785:0x3ffb1f40 0x400dc95b:0x3ffb1f60 0x400e6f39:0x3ffb1fb0 0x40089369:0x3ffb1fd0

Rebooting...
----------------

Edzelf, 2020-08-17:
D: pin_tft_scl set to 18
D: pin_tft_sda set to 23
Pin 18 and 23 are also used for:
D: pin_spi_sck set to 18
D: pin_spi_mosi set to 23
So there are pin conflicts.

Aleks-Ale, 2020-08-17:
thank you for answer!
So, this table is not correct?
https://yadi.sk/i/7QQwKDSzkX2Xpg
Esp32_radio.ino
// Wiring. Note that this is just an example.  Pins (except 18,19 and 23 of the SPI interface)
// can be configured in the config page of the web interface.
// ESP32dev Signal  Wired to LCD        Wired to VS1053      SDCARD   Wired to the rest
// -------- ------  --------------      -------------------  ------   ---------------
// GPIO32           -                   pin 1 XDCS            -       -
// GPIO5            -                   pin 2 XCS             -       -
// GPIO4            -                   pin 4 DREQ            -       -
// GPIO2            pin 3 D/C or A0     -                     -       -
// GPIO22           -                   -                     CS      -
// GPIO16   RXD2    -                   -                     -       TX of NEXTION (if in use)
// GPIO17   TXD2    -                   -                     -       RX of NEXTION (if in use)
// GPIO18   SCK     pin 5 CLK or SCK    pin 5 SCK             CLK     -
// GPIO19   MISO    -                   pin 7 MISO            MISO    -
// GPIO23   MOSI    pin 4 DIN or SDA    pin 6 MOSI            MOSI    -
// GPIO15           pin 2 CS            -                     -       -
// GPI03    RXD0    -                   -                     -       Reserved serial input
// GPIO1    TXD0    -                   -                     -       Reserved serial output
// GPIO34   -       -                   -                     -       Optional pull-up resistor
// GPIO35   -       -                   -                     -       Infrared receiver VS1838B
// GPIO25   -       -                   -                     -       Rotary encoder CLK
// GPIO26   -       -                   -                     -       Rotary encoder DT
// GPIO27   -       -                   -                     -       Rotary encoder SW
// -------  ------  ---------------     -------------------  ------   ----------------

Esp32_radio_init.ino
----------------------------
  preferences.putString ( "pin_vs_cs",   "5                                      # GPIO Pin number for VS1053 CS" ) ;
  preferences.putString ( "pin_vs_dcs",  "16                                     # GPIO Pin number for VS1053 DCS" ) ;
  preferences.putString ( "pin_vs_dreq", "4                                      # GPIO Pin number for VS1053 DREQ" ) ;
  //
  preferences.putString ( "pin_enc_clk", "25                                     # GPIO Pin number for rotary encoder CLK" ) ;
  preferences.putString ( "pin_enc_dt",  "26                                     # GPIO Pin number for rotary encoder DT" ) ;
  preferences.putString ( "pin_enc_sw",  "27                                     # GPIO Pin number for rotary encoder SW" ) ;
  //
  preferences.putString ( "pin_tft_cs",  "15                                     # GPIO Pin number for TFT CS" ) ;
  preferences.putString ( "pin_tft_dc",  "2                                      # GPIO Pin number for TFT DC" ) ;
  //
  preferences.putString ( "pin_sd_cs",   "22                                     # GPIO Pin number for SD card CS" ) ;

Please, which changes is necessary for work without conflict?
Thank you!

Edzelf, 2020-08-17:
You don't need the Esp32_radio_int.ino anymore, but you can use it to clear the NVS.  Then you can change the setting in the config page of the webinterface.
See the documentation.

Aleks-Ale, 2020-08-17:
My access point "brg" gave a ip address 192.168.1.115.
D: Connect to WiFi
D: Connected to brg
D: IP = 192.168.1.115
But unfortunately, I cannot access the web page at this address.

Aleks-Ale, 2020-08-17:
> D: pin_tft_scl set to 18
> D: pin_tft_sda set to 23
> Pin 18 and 23 are also used for:
> D: pin_spi_sck set to 18
> D: pin_spi_mosi set to 23
> So there are pin conflicts.
-----------------------------------------
I changed the display connection. 
The display is working. But the reboot is still there

Esp32_radio.ino
-----------------
    { "pin_tft_cs",    &ini_block.tft_cs_pin,       15 }, // Display SPI version
    { "pin_tft_dc",    &ini_block.tft_dc_pin,        2 }, // Display SPI version
--->    //{ "pin_tft_scl",   &ini_block.tft_scl_pin,      18 }, // Display I2C version
    { "pin_tft_scl",   &ini_block.tft_scl_pin,      16 }, // Display I2C version
--->    //{ "pin_tft_sda",   &ini_block.tft_sda_pin,      23 }, // Display I2C version
    { "pin_tft_sda",   &ini_block.tft_sda_pin,      21 }, // Display I2C version
    { "pin_tft_bl",    &ini_block.tft_bl_pin,       -1 }, // Display backlight
    { "pin_tft_blx",   &ini_block.tft_blx_pin,      -1 }, // Display backlight (inversed logic)

    { "pin_spi_sck",   &ini_block.spi_sck_pin,      18 },
    { "pin_spi_miso",  &ini_block.spi_miso_pin,     19 },
    { "pin_spi_mosi",  &ini_block.spi_mosi_pin,     23 },
....
-->  //setup_CH376() ;                                        // Init CH376 if configured
-----------------

bluetft.h
-----------------
bool dsp_begin()
{
  tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
                              ini_block.tft_dc_pin, 
                              ini_block.tft_sda_pin,
                              ini_block.tft_scl_pin, -1 ) ;        // Create an instant for TFT
-----------------

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 09:40:00 GMT.  Free memory 288144
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 16
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
[... 89 lines cut ...]
D: Connected to brg
D: IP = 192.168.1.115
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 16:33:13
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e1e77  PS      : 0x00060430  A0      : 0x800d2798  A1      : 0x3ffb1f20  
A2      : 0x00000000  A3      : 0x3f402896  A4      : 0x3ffb1f28  A5      : 0x3ffc8fa8  
A6      : 0x3ffc8fbc  A7      : 0x00000000  A8      : 0x800e6244  A9      : 0x3ffb1e50  
A10     : 0x3ffb1eac  A11     : 0x3f402896  A12     : 0x00000001  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e1e77:0x3ffb1f20 0x400d2795:0x3ffb1f40 0x400dc96f:0x3ffb1f60 0x400e6f69:0x3ffb1fb0 0x40089369:0x3ffb1fd0

Rebooting...

Rainer-G, 2020-08-17:
Do you have a rotary encoder and/or an infrared receiver connected?
If not, I would set the unused pins to -1 to disable them. 
Pin 35, it has no pull-up and floating pins with interrupt routines attached may trigger spurious interrupts.
Noise on pins for rotary encoder may also cause trouble!
The rest seems to work, display, network, vs1053, connection to radio station
Cheers
Rainer

Von: Aleks-Ale
Gesendet: Montag, 17. August 2020 16:48
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: Re: [Edzelf/ESP32-Radio] no first start. reboot only (#363)

D: pin_tft_scl set to 18
D: pin_tft_sda set to 23
Pin 18 and 23 are also used for:
D: pin_spi_sck set to 18
D: pin_spi_mosi set to 23
So there are pin conflicts.

I changed the display connection.
The display is working. But the reboot is still there
Esp32_radio.ino
{ "pin_tft_cs",    &ini_block.tft_cs_pin,       15 }, // Display SPI version
{ "pin_tft_dc",    &ini_block.tft_dc_pin,        2 }, // Display SPI version
---> //{ "pin_tft_scl", &ini_block.tft_scl_pin, 18 }, // Display I2C version
{ "pin_tft_scl", &ini_block.tft_scl_pin, 16 }, // Display I2C version
---> //{ "pin_tft_sda", &ini_block.tft_sda_pin, 23 }, // Display I2C version
{ "pin_tft_sda", &ini_block.tft_sda_pin, 21 }, // Display I2C version
{ "pin_tft_bl", &ini_block.tft_bl_pin, -1 }, // Display backlight
{ "pin_tft_blx", &ini_block.tft_blx_pin, -1 }, // Display backlight (inversed logic)
{ "pin_spi_sck",   &ini_block.spi_sck_pin,      18 },
{ "pin_spi_miso",  &ini_block.spi_miso_pin,     19 },
{ "pin_spi_mosi",  &ini_block.spi_mosi_pin,     23 },
....
--> //setup_CH376() ; // Init CH376 if configured
bluetft.h
bool dsp_begin()
{
tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
ini_block.tft_dc_pin,
ini_block.tft_sda_pin,
ini_block.tft_scl_pin, -1 ) ; // Create an instant for TFT
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
D: Starting ESP32-radio running on CPU 1 at 240 MHz. Version Tue, 14 Jul 2020 09:40:00 GMT. Free memory 288144
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 16
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
[... 92 lines cut ...]
D: endFillByte is 0
D: Connect to WiFi
D: Connected to brg
D: IP = 192.168.1.115
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 16:33:13
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x400e1e77 PS : 0x00060430 A0 : 0x800d2798 A1 : 0x3ffb1f20
A2 : 0x00000000 A3 : 0x3f402896 A4 : 0x3ffb1f28 A5 : 0x3ffc8fa8
A6 : 0x3ffc8fbc A7 : 0x00000000 A8 : 0x800e6244 A9 : 0x3ffb1e50
A10 : 0x3ffb1eac A11 : 0x3f402896 A12 : 0x00000001 A13 : 0x0000ff00
A14 : 0x00ff0000 A15 : 0xff000000 SAR : 0x0000000a EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000038 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0xffffffff
Backtrace: 0x400e1e77:0x3ffb1f20 0x400d2795:0x3ffb1f40 0x400dc96f:0x3ffb1f60 0x400e6f69:0x3ffb1fb0 0x40089369:0x3ffb1fd0
Rebooting...

Aleks-Ale, 2020-08-17:
> Do you have a rotary encoder and/or an infrared receiver connected? If not, I would set the unused pins to -1 to disable them. Pin 35, it has no pull-up and floating pins with interrupt routines attached may trigger spurious interrupts.

Thanks for the good advice! 
```
  struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",        &ini_block.ir_pin,           -1 },
```
System started successfully.
I have a rotary encoder but unfortunately it doesn't work. 
Rotation and pressing have no effect.

```
  struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",             &ini_block.ir_pin,           -1 },
    { "pin_enc_clk",   &ini_block.enc_clk_pin,      25 },
    { "pin_enc_dt",    &ini_block.enc_dt_pin,       26 },
    { "pin_enc_sw",    &ini_block.enc_sw_pin,       27 },
```

Rainer-G, 2020-08-18:
Did you check the wiring?

For a quick test, press the encoder’s switch while rebooting and do not release until this part of the log appears in the serial monitor: 
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
Pin 27 should now be LOW and this proves that at least pin 27 and GND are connected.

If so, check in the log once the radio is playing a song, then watch the serial monitor for any encoder related debug messages

Von: Aleks-Ale
Gesendet: Montag, 17. August 2020 23:56
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] no first start. reboot only (#363)

Do you have a rotary encoder and/or an infrared receiver connected? If not, I would set the unused pins to -1 to disable them. Pin 35, it has no pull-up and floating pins with interrupt routines attached may trigger spurious interrupts.
Thanks for the good advice!
  struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",        &ini_block.ir_pin,           -1 },
System started successfully.
I have a rotary encoder but unfortunately it doesn't work.
Rotation and pressing have no effect.
  struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",             &ini_block.ir_pin,           -1 },
    { "pin_enc_clk",   &ini_block.enc_clk_pin,      25 },
    { "pin_enc_dt",    &ini_block.enc_dt_pin,       26 },
    { "pin_enc_sw",    &ini_block.enc_sw_pin,       27 },

Aleks-Ale, 2020-08-18:
> If so, check in the log once the radio is playing a song,

That's my fault! I commented out the encoder line to run in light configuration and forgot about it
```
  //chk_enc() ;                                       // Check rotary encoder functions
  //check_CH376() ;                                   // Check Flashdrive insert/remove
```
The encoder is working now. I need to learn to use it. :)))
Double-tap to select a station activates the Mute. 
For example, I make five attempts to double-tap and three times this turns on the mute mode 2 times allows me to change the station. Contact bounce?
I will open new topic with new questions.
Thank you for first start!

Aleks-Ale, 2020-08-19:
Thanks for all

closed 2020-08-19 by Aleks-Ale