Edzelf/ESP32-Radio #364: A few questions after the first start.

Aleks-Ale, 2020-08-18:
1. How to listen https:// radio ?
https://online.radiokarnaval.ru/karnaval
https://pub0301.101.ru:8443/stream/trust/mp3/128/264
2. Is it possible to change the encoder logic?
- I don't need volume change. Always 100%.
- Turning the encoder only selects the station
- Pressing the encoder confirms the station selection.
3. How to change font size for station name?

Edzelf, 2020-08-19:
1 - https: is not supported.
2 - It is not that difficult to change the behaviour of the encoder.  But try to do it yourself.
3 - The font size is also easy to change.  But take care of the position on the screen, specially for the time display.

Aleks-Ale, 2020-08-19:
Thank you for answer

closed 2020-08-19 by Aleks-Ale