Edzelf/ESP32-Radio #366: Something looks broken in the latest version, doesn't try to connect to wifi

greppp, 2020-09-06:
Hi,

I worked a bit on this project this summer, put it aside and I am back at it. I decided to download the lastest version of the code and it's been 2 days I can't get it to work.

This is what I get with the fresh code, as you can see it stops after the CH376 thing and nothing ever happen after.
[image]

I pulled my files from this summer and this is what I get, as you can see now it tries to connect to the WiFi.
[image]

Edzelf, 2020-09-07:
Try to remove the CH176 support by commenting line 167:
//#define CH376

greppp, 2020-09-22:
Hi @Edzelf, thanks. Shouldn’t be this commented by default? Or at least it would be nice for it to be in the doc when you have some time.
Haven’t had time to try, will try this week, i mostly worked with my old version until now.

fiegeman, 2020-10-25:
Thank you, commented it out and it works perfectly. Thank you for all your assistance.