Edzelf/ESP32-Radio #367: output the station name from defaultprefs. h to TFT 1602

avulyanichev, 2020-09-07:
//I found the readhostfrompref (); function in the sketch that outputs a string with the current information by the preset number (currentpreset).
//selecting the line after #
String n_presset ; // I added a variable to display the station name.
String n_presset = readhostfrompref(cur rentpreset); // 
uint8_t n_simvola = n_presset.indexOf("# ");// finding a location "#"
dsp_println (n_presset.substring (n_simvola + 2)); // +2 this means that the '#' character with a space does not appear on the screen

//tell me how to insert this data in LCD1602.h in  -----   dline[1].str = (tftdata[2].str);                     // Local copy

greppp, 2020-09-14:
I am trying to achieve kind of the same thing (https://github.com/Edzelf/ESP32-Radio/issues/370) so I am interested to see how it goes!

avulyanichev, 2020-09-16:
http://www.cqham.ru/forum/showthread.php?40897-%CF%F0%EE%E5%EA%F2-%E2%FB%F5%EE%E4%ED%EE%E3%EE-%E4%ED%FF-%C8%ED%F2%E5%F0%ED%E5%F2-%F0%E0%E4%E8%EE-%ED%E0-ESP32-VS1053-ILI9341/page2
На экран 320x480 сделано

greppp, 2020-09-22:
Great, thanks, had a quick look at it his week-end and didn’t success but I will try again this week.

avulyanichev, 2020-10-08:
как результаты? У меня пока не получилось.

greppp, 2020-10-19:
Hi avulyanichev, no not yet.

So far no succes :( I really would like this to work, for me a radio only displays the radio station name, the rest is clutter I find. I am a beginner with coding so I am afraid I won't be the one finding how to do that. Let me know if you find anything.

I would also like the encoder to act like the buttons, by that i mean when a turn it it switches to the next station right away and not ask for confirmation.

I'll keep searching.`

avulyanichev, 2020-10-19:
Я занят был, скоро буду опять искать решение вывода названия станции. Надеюсь решение найдем.