Edzelf/ESP32-Radio #371: Problem starting

trombose009, 2020-09-21:
Hello, 

I have a new problem during boot process.

D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added fritzboxdampf to list of networks
D: Added dampfmaschine2 to list of networks
D: End adding networks

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 09:40:00 GMT.  Free memory 247920
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added fritzboxdampf to list of networks
D: Added dampfmaschine2 to list of networks
D: End adding networks

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 14 Jul 2020 09:40:00 GMT.  Free memory 247920
D: Display type is BLUETFT
.......

During the last days, I tried to connect the radio to my wifi. I added 2 wifi credentials to the ESP32_radio_init.ino, uploaded it and the upload the esp32_radio.ino.
After that, in the log, the two SSIDs shwon up but it just did not connect to wifi. 
Today, I wanted to give it another try but now I have some error during boot process. 
Does anyone has an idea what'S the problem?

greetings

Edzelf, 2020-09-21:
"brownout" means that the voltage of the power source is lower than a certain limit.  In other words: the power source is too weak.

trombose009, 2020-09-21:
ok. that was easy :-P

but now, the first problem. It does not connect to the wifi:

D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added fritzboxdampf to list of networks
D: Added dampfmaschine2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - dampfmaschine2            Signal: -77 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - fritzboxdampf             Signal: -84 dBm, Encryption WPA_WPA2_PSK, Acceptable
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Start CH376

here it stops.

Edzelf, 2020-09-21:
Remove the definition of CH376 and SDCARD if you don't have it.  Around line 168.

trombose009, 2020-09-21:
Thank you so much. It's great to get answers also to such easy questions.

Now I run into a wifi problem:

D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
E (5722) wifi: esp_wifi_scan_start 1245 wifi not start
E (5722) wifi: esp_wifi_disconnect 1153 wifi not start
E (5722) wifi: esp_wifi_scan_start 1245 wifi not start
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is enabled
D: STOP requested
D: Song stopped correctly after 0 msec

I just did some reacherches in the internet and figured out, that it might be a bug in the wifi library. 
But maybe someone does know the issue and can help.

trombose009, 2020-09-23:
Edzelf solved my starting problems. So I dio another case for the wifi thing

closed 2020-09-23 by trombose009