Edzelf/ESP32-Radio #373: No Sound

joeck, 2020-09-21:
I can control the radio with the web interface but I cant get any sound from the 3.5mm output of the VS1003/VS1053 (label on Controlller).

I'm using PCB made by nakedninja. Display (1.8TFT SP 128*160 V1.1) currently disconnected

Logs:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 288308
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 27 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Start display
D: Create list with acceptable WiFi networks
D: Added Bluewin to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 6
D:  1 - Bluewin                   Signal: -58 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - Bluewin                   Signal: -73 dBm, Encryption WPA2_PSK, Acceptable
D:  3 - Sunrise_2.4GHz_279CD0     Signal: -85 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - mur-61097                 Signal: -91 dBm, Encryption WPA2_PSK, 
D:  5 - Maral                     Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D:  6 - UPC Wi-Free               Signal: -94 dBm, Encryption ????, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter -1
D: Command: clk_server with parameter pool.ntp.org
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter energycharts.ice.infomaniak.ch/energycharts-high.mp3
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 SPI error. SB:012C R1:0000 R2:0000
D: VS1053 SPI error. SB:0258 R1:0000 R2:0000
D: VS1053 SPI error. SB:0384 R1:0000 R2:0000
D: VS1053 SPI error. SB:04B0 R1:0000 R2:0000
D: VS1053 SPI error. SB:05DC R1:0000 R2:0000
D: VS1053 SPI error. SB:0708 R1:0000 R2:0000
D: VS1053 SPI error. SB:0834 R1:0000 R2:0000
D: VS1053 SPI error. SB:0960 R1:0000 R2:0000
D: VS1053 SPI error. SB:0A8C R1:0000 R2:0000
D: VS1053 SPI error. SB:0BB8 R1:0000 R2:0000
D: VS1053 SPI error. SB:0CE4 R1:0000 R2:0000
D: VS1053 SPI error. SB:0E10 R1:0000 R2:0000
D: VS1053 SPI error. SB:0F3C R1:0000 R2:0000
D: VS1053 SPI error. SB:1068 R1:0000 R2:0000
D: VS1053 SPI error. SB:1194 R1:0000 R2:0000
D: VS1053 SPI error. SB:12C0 R1:0000 R2:0000
D: VS1053 SPI error. SB:13EC R1:0000 R2:0000
D: VS1053 SPI error. SB:1518 R1:0000 R2:0000
D: VS1053 SPI error. SB:1644 R1:0000 R2:0000
D: VS1053 SPI error. SB:1770 R1:0000 R2:0000
D: This is not a VS1053, but a VS1001 instead!
D: Connect to WiFi
D: Try WiFi Bluewin
D: Connected to Bluewin
D: IP = 192.168.1.144
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 15:28:45
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh14
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 74
D: Song stopped correctly after 0 msec
D: Duration mp3loop 201
D: Metadata block 48 bytes
D: Streamtitle found, 47 bytes
D: StreamTitle='NPO Radio 1 - Nieuws en Co - NTR';

Edzelf, 2020-09-22:
Your chip is a VS1001, not a VS1053.  I can not test this chip.

AlohaFromHell76, 2020-09-23:
Exactly the same here. I also use the nakedninja PCB.16:22:47.197 -> ets Jun  8 2016 00:22:57
16:22:47.197 -> 
16:22:47.197 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:22:47.197 -> configsip: 0, SPIWP:0xee
16:22:47.197 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:22:47.197 -> mode:DIO, clock div:1
16:22:47.197 -> load:0x3fff0018,len:4
16:22:47.197 -> load:0x3fff001c,len:1216
16:22:47.197 -> ho 0 tail 12 room 4
16:22:47.197 -> load:0x40078000,len:9720
16:22:47.197 -> ho 0 tail 12 room 4
16:22:47.197 -> load:0x40080400,len:6352
16:22:47.197 -> entry 0x400806b8
16:22:47.517 -> 
16:22:47.517 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 17 Sep 2020 12:50:00 GMT.  Free memory 294308
16:22:47.517 -> D: Display type is DUMMYTFT
16:22:47.517 -> D: Partition nvs found, 20480 bytes
16:22:47.517 -> D: Read 36 keys from NVS
16:22:47.517 -> D: pin_ir set to -1
16:22:47.517 -> D: pin_enc_clk set to -1
16:22:47.517 -> D: pin_enc_dt set to -1
16:22:47.517 -> D: pin_enc_sw set to -1
16:22:47.517 -> D: pin_tft_cs set to 15
16:22:47.517 -> D: pin_tft_dc set to 2
16:22:47.517 -> D: pin_tft_scl set to -1
16:22:47.564 -> D: pin_tft_sda set to -1
16:22:47.564 -> D: pin_tft_bl set to -1
16:22:47.564 -> D: pin_tft_blx set to -1
16:22:47.564 -> D: pin_sd_cs set to -1
16:22:47.564 -> D: pin_ch376_cs set to -1
16:22:47.564 -> D: pin_ch376_int set to -1
16:22:47.564 -> D: pin_vs_cs set to 5
16:22:47.564 -> D: pin_vs_dcs set to 16
16:22:47.564 -> D: pin_vs_dreq set to 4
16:22:47.564 -> D: pin_shutdown set to -1
16:22:47.564 -> D: pin_shutdownx set to -1
16:22:47.564 -> D: pin_spi_sck set to 18
16:22:47.564 -> D: pin_spi_miso set to 19
16:22:47.564 -> D: pin_spi_mosi set to 23
16:22:47.564 -> D: GPIO0 is HIGH
16:22:47.564 -> D: GPIO2 is LOW, probably no PULL-UP
16:22:47.611 -> D: GPIO4 is HIGH
16:22:47.611 -> D: GPIO5 is HIGH
16:22:47.611 -> D: GPIO12 is HIGH
16:22:47.611 -> D: GPIO13 is HIGH
16:22:47.611 -> D: GPIO14 is HIGH
16:22:47.657 -> D: GPIO15 is HIGH
16:22:47.657 -> D: GPIO16 is HIGH
16:22:47.657 -> D: GPIO17 is HIGH
16:22:47.657 -> D: GPIO18 is HIGH
16:22:47.703 -> D: GPIO19 is HIGH
16:22:47.703 -> D: GPIO21 is HIGH
16:22:47.703 -> D: GPIO22 is HIGH
16:22:47.703 -> D: GPIO23 is HIGH
16:22:47.703 -> D: GPIO25 is HIGH
16:22:47.751 -> D: GPIO26 is HIGH
16:22:47.751 -> D: GPIO27 is HIGH
16:22:47.751 -> D: GPIO32 is HIGH
16:22:47.751 -> D: GPIO33 is HIGH
16:22:47.751 -> D: GPIO34 is LOW, probably no PULL-UP
16:22:47.797 -> D: GPIO35 is LOW, probably no PULL-UP
16:22:47.797 -> D: GPIO39 is LOW, probably no PULL-UP
16:22:47.797 -> D: gpio_00 will execute uppreset = 1
16:22:47.797 -> D: gpio_12 will execute upvolume = 2
16:22:47.797 -> D: gpio_13 will execute downvolume = 2
16:22:47.797 -> D: Start display
16:22:47.797 -> D: Create list with acceptable WiFi networks
16:22:47.797 -> D: Added DerSchwarzeKanal to list of networks
16:22:47.797 -> D: Added SSID2 to list of networks
16:22:47.797 -> D: End adding networks
[... 122 lines cut ...]
16:22:57.600 -> D: Duration mptopic PFD: Publish to topic PFFA4AE30/icy/name : 181.FM Beatles
16:22:57.600 -> 
16:22:57.694 -> D: Publish to topic PFFA4AE30/playing : 1
16:22:58.676 -> D: Metadata block 48 bytes
16:22:58.676 -> D: Streamtitle found, 34 bytes
16:22:58.676 -> D: StreamTitle='Beatles - Hey Jude ';
16:22:58.676 -> D: Command client available
16:22:58.676 -> D: handlehttp started
16:22:58.676 -> D: Publish to topic PFFA4AE30/icy/streamtitle : Beatles - Hey Jude 
16:22:58.676 -> D: Get command is: status
16:22:58.676 -> D: Send reply for status
16:22:58.723 -> D: Command: status with parameter 0
16:22:59.000 -> D: Command client available
16:22:59.000 -> D: handlehttp started
16:22:59.000 -> D: HTTP input shorter than expected
16:22:59.695 -> D: Command client available
16:22:59.695 -> D: handlehttp started
16:22:59.742 -> D: HTTP input shorter than expected
16:23:03.833 -> D: Command client available
16:23:03.833 -> D: handlehttp started
16:23:03.833 -> D: Get command is: status
16:23:03.833 -> D: Send reply for status
16:23:03.833 -> D: Command: status with parameter 0

I replaced the VS1053 with another  ... still no sound. How can i get a real VS1053? They are all labeled with VS1003/1053?

Edzelf, 2020-09-23:
Your chip is still a VS1001, not a VS1053. I can not test this chip.