Edzelf/ESP32-Radio #374: Compiler error

udope, 2020-09-21:
I can't compile the sketch. The error is:

Esp32_radio_compiliert:1981:19: error: variable or field 'setdatamode' declared void
 void setdatamode ( datamode_t newmode )

                   ^
Esp32_radio_compiliert:1981:19: error: 'datamode_t' was not declared in this scope

Can you help me, please?

Edzelf, 2020-09-22:
I guess your source file is corrupt.  "datamode_t" is defined at line 366.
The function "setdatamode()" is at line 1979, not at line 1981.

udope, 2020-09-22:
First of all: **Thankyou** for your quick return.

I made a new download, "datamode_t" ist defined in 366 and "setdatamode()" is in line 1979 in the new file, but the result is the same:

Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi/BT), QIO, 40MHz, 4MB (32Mb), 921600, None"

Esp32_radio:1979:19: error: variable or field 'setdatamode' declared void

 void setdatamode ( datamode_t newmode )

                   ^

Esp32_radio:1979:19: error: 'datamode_t' was not declared in this scope

exit status 1
variable or field 'setdatamode' declared void

Thankyou for your answer!

udope, 2020-09-24:
Hi!
Solved - compiled on a new, fresh Arduino installtion - worked.