Edzelf/ESP32-Radio #380: LCD44780 compatible LCD1602, is it possible to add to support

xiaolaba, 2020-09-28:
hello, we had try to include to support LCD 44780 compatible LCD1602, but somehow didn't work out.
any idea how to work out this, appreciate your help

`
//#define LCD1602I2C                   // LCD 1602 display with I2C backpack
//#define LCD2004I2C                   // LCD 2004 display with I2C backpack
//#define ILI9341                      // ILI9341 240*320
//#define NEXTION                      // Nextion display. Uses UART 2 (pin 16 and 17)
#define LCD1602                   // LCD 1602 display
//

// include the library code:  // standard lcd 44780 compatible LCD1602
#include <LiquidCrystal.h>
`

Edzelf, 2020-09-28:
Does it have a I2C interface?  Do you know the I2C address?

xiaolaba, 2020-09-29:
sorry about no very clear description.
we use the Arduino LCD library usually with such LCD, so it should be a parallel type of interface, should be not I2C type.

the testing code below and display done properly,
`
// include the library code:
#include <LiquidCrystal.h>

const int rs = 2, en = 3, d4 = 4, d5 = 5, d6 = 6, d7 = 7;
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);

void setup() {
  // set up the LCD's number of columns and rows:
  lcd.begin(16, 2);
  // Print a message to the LCD.
  lcd.print("hello, world!");
}
`

Edzelf, 2020-09-29:
Are you sure there are enough pins to control the VS1053 and the LCD?