Edzelf/ESP32-Radio #382: Guru Meditation Error: Core  0

h1aji, 2020-10-04:
Hi @Edzelf I compiled with LCD2004 and after few days I am getting this error:

> D: Rotary encoder is disabled (-1/-1/-1)
> D: Sync TOD
> D: Sync TOD, new value is 23:00:09
> D: STOP requested
> D: New preset/file requested (1/0) from ice912.echo.msk.ru:9120/24.aac
> D: Connect to new host ice912.echo.msk.ru:9120/24.aac
> D: Connect to ice912.echo.msk.ru on port 9120, extension /24.aac
> D: Song stopped correctly after 0 msec
> Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x4000c3f0  PS      : 0x00060b30  A0      : 0x80127804  A1      : 0x3ffd7eb0  
> A2      : 0x3ffd82f8  A3      : 0x4008aba4  A4      : 0x00000002  A5      : 0x00000000  
> A6      : 0x00000020  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x4008aba4  
> A10     : 0x00000030  A11     : 0x00000000  A12     : 0x3ffb85c8  A13     : 0x00000000  
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000003  
> EXCVADDR: 0x4008aba4  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x4000c3f0:0x3ffd7eb0 0x40127801:0x3ffd7ed0 0x4012058a:0x3ffd7f00 0x4011d075:0x3ffd8210 0x400d2717:0x3ffd82d0 0x400d4646:0x3ffd8330 0x400d4676:0x3ffd8350 0x4008aba9:0x3ffd8370
> 
> Rebooting...

But when I connect just ESP32 module to my PC using microusb it seems to be booting fine, web interface is loading.
I have to compile sketch with DUMMYTFT to make it working. I am not sure if its because of LCD2004 module

Edzelf, 2020-10-04:
aac is only supported by VS1053.  Are you sure you have one?

h1aji, 2020-10-05:
[image]

I am using this one. And generally this stream and other AAC are working fine, its just mb after week of use it fails with this error.

Edzelf, 2020-10-05:
Sorry, I cannot reproduce the error:
```
D: Get command is: station=ice912.echo.msk.ru:9120/24.aac
D: Send reply for station=ice912.echo.msk.ru:9120/24.aac
D: Command: station with parameter ice912.echo.msk.ru:9120/24.aac
D: STOP requested
D: Stopping client
D: Connect to new host ice912.echo.msk.ru:9120/24.aac
D: Connect to ice912.echo.msk.ru on port 9120, extension /24.aac
D: Song stopped correctly after 20 msec
D: Connected to server
D: Switch to HEADER
D: Headerline: Server: nginx/1.15.7
D: Headerline: Content-Type: audio/aac
D: audio/aac seen.
D: Headerline: Transfer-Encoding: chunked
D: Headerline: Connection: close
D: Headerline: icy-br: 24
D: Headerline: ice-audio-info: channels=2;samplerate=48000;bitrate=24
D: Headerline: icy-name: /24.aac
D: Headerline: icy-pub: 1
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint: 8192
D: Switch to DATA, bitrate is 24, metaint is 8192
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
```
Use the "ESP exception decoder"-plug-in to see where the exception occurs.

h1aji, 2020-10-18:
Got this:
```
PC: 0x4000c3f0
EXCVADDR: 0x4008aba4: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 142

Decoding stack results
0x401277fd: __ssprint_r at ../../../.././newlib/libc/stdio/vfprintf.c line 342
0x40120586: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1774
0x4011d071: sprintf at ../../../.././newlib/libc/stdio/sprintf.c line 644
0x400d2713: displaytime(char const*, unsigned short) at /tmp/arduino_build_234881/sketch/LCD2004.h line 476
0x400d4642: handle_spec() at /home/hmurad/git/ESP32-Radio/Esp32_radio/Esp32_radio.ino line 5525
0x400d4672: spftask(void*) at /home/hmurad/git/ESP32-Radio/Esp32_radio/Esp32_radio.ino line 5556
0x4008aba9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

h1aji, 2020-10-18:
as soon as i cleared displayvolume block
```void displaytime ( const char* str, uint16_t color )
{
}
```
it started working
Any chance to get this fixed? Thanks

Edzelf, 2020-10-18:
Fixed.

h1aji, 2020-10-19:
thank you @Edzelf

h1aji, 2020-10-19:
unfortunately now I see this:
```
PC: 0x40148f78: i2c_cmd_link_delete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2c.c line 962
EXCVADDR: 0x00323380

Decoding stack results
0x40148f78: i2c_cmd_link_delete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2c.c line 962
0x400d23b3: LCD2004::scommand(unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 180
0x400d23c7: LCD2004::strobe(unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 153
0x400d23fb: LCD2004::swrite(unsigned char, unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 130
0x400d2420: LCD2004::write_cmd(unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 142
0x400d245b: LCD2004::scursor(unsigned char, unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 206
0x400d24dd: dsp_update_line(unsigned char) at /tmp/arduino_build_869484/sketch/LCD2004.h line 357
0x400d2766: displaytime(char const*, unsigned short) at /tmp/arduino_build_869484/sketch/LCD2004.h line 494
0x400d46a2: handle_spec() at /home/hmurad/git/ESP32-Radio/Esp32_radio/Esp32_radio.ino line 5529
0x400d46d2: spftask(void*) at /home/hmurad/git/ESP32-Radio/Esp32_radio/Esp32_radio.ino line 5560
0x4008aba9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

Edzelf, 2020-10-19:
Please show full logging.

h1aji, 2020-10-19:
@Edzelf same as in this post https://github.com/Edzelf/ESP32-Radio/issues/393#issuecomment-711871731
In my case SDA and SCL are 13 and 14. I think this problem is with **displaytime**
As soon as i comment it the radio starts working fine

Edzelf, 2020-10-19:
I will try to reproduce the error.

Edzelf, 2020-10-19:
Fixed.

hevet, 2020-10-19:
Cool, looks like it's working fine.

Unfortunately, the volume level has disappeared.
In line 475 it is, 

dline[3].str += "\x7F" ;                      // Add block character

and from what I can see it's a blank character, the dash is x2D.

h1aji, 2020-10-19:
Looks fine now. Thanks

closed 2020-10-19 by h1aji