Edzelf/ESP32-Radio #384: WIFI connection

yakuza66, 2020-10-07:
How to connect to WiFi, where you must enter SSID name, USER name and PASSWORD ?
Network properties:
SSID
WPA2-Enterprise
AES/TKIP
PEAP
USER NAME / PASSWORD

Edzelf, 2020-10-07:
The radio starts in AP mode.  Make a connection to 192.168.4.1 and enter your WiFi credentials in the config page of the web interface.

yakuza66, 2020-10-07:
I did it for my home WiFi, where i need to enter just SSID and PASSWORD ( eg. wifi_00 =  SSID/password) and wors fine. But in my office I need to enter SSID, USERNAME and PASSWORD. In what form do I need enter this WiFi credentials in config page line ?

Edzelf, 2020-10-07:
I think it not possible to connect an ESP32 to such a network.  If you can find a solution, let me know.

macikm, 2020-10-07:
> I think it not possible to connect an ESP32 to such a network.  If you can find a solution, let me know.

not with WifiManager, but maybe a solution …

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEnterprise/WiFiClientEnterprise.ino

Edzelf, 2020-10-07:
I do not have the possibility to test it.