Edzelf/ESP32-Radio #386: vs1003 is too hot

clear-sky, 2020-10-07:
hi all
i have a fake vs1053 board that has vs1003 chip
i read all comments and could use it (commented line 921 as below)
//    okay = false ;
i can hear mp3 stations but chip vs1003 is too hot and i can not stay my finger on it more than 10 seconds

is that my problem or this is normal?

[image]

clear-sky, 2020-10-21:
to dear vs1053 owners
do you have this problem?

Edzelf, 2020-10-22:
Mine (VS1053) is 22.1 degrees Centigrade.

clear-sky, 2020-10-22:
thanks dear ed 
you are very kind

clear-sky, 2020-10-24:
hi all
found the problem
i had connected earphone output to pam8403 amplifier
in fact i had connected earphone  ground to system ground 
earphone ground is not normal ground and should not be connect to system ground 
if you make this mistake high current passes through vs1003 and cause chip overheat !
just disconnected ground between vs1003 module and pam 8403 !
now my vs1003 is cold !

closed 2020-10-24 by clear-sky

Edzelf, 2020-10-24:
Yes, you have to use some kind of isolation like double powersupplies or transformers.