Edzelf/ESP32-Radio #389: Unreliabe

FrankBoesing, 2020-10-16:
There are many dropouts and very often streams do not start at all.

The power is supplied by a 3A USB power supply, this is not the reason.

I have a reliable 50MBit internet connection.
On the PC I have no such problems.

I use the Heltec board with OLED (wifi Kit32)

What could be the reason?

Edzelf, 2020-10-16:
Bad luck I guess.....

FrankBoesing, 2020-10-16:
Does it make sense to try an other board?

Edzelf, 2020-10-16:
Yes, I would try that.  If that fails, try a different Internet connection.

FrankBoesing, 2020-10-16:
...I read in an other issue the hint to disable the updates. So I added a 
//#define OTA and 
```
#if OTA
#include <ArduinoOTA.h>
#include <Update.h>
#endif
```

and added this switch to all parts in the ino which use these libs.

:) Works reliable now.
I guess there is either a bug somewhere (or a timing/memory problem)?

I can add a pullrequest, if you want me to.

FrankBoesing, 2020-10-24:
I added a short wire as Antenna.
Way better now.
Thanks.

closed 2020-10-24 by FrankBoesing